waveform80/picamera #481: Access to vc.ril.isp?

nsimon1, 2018-04-10:
Hello, 

I would like to know if there is access through mmalobj to the vc.ril.isp component? I want to use it to convert a YUVY video stream to I420 and then pass that along to the encoder. 

My end goal is something like this:
```
MMALRawCam() [YUVY output stream] --> vc.ril.isp [I420 output] --
--> MMALVideoEncoder() [H.264 output] --> stdout
```

6by9, 2018-04-11:
MMALISPResizer already exists as an object and should do what you want
https://picamera.readthedocs.io/en/latest/api_mmalobj.html#picamera.mmalobj.MMALISPResizer

nsimon1, 2018-04-11:
@6by9 Thank you very much for your reply, that is great! 

I started off trying to make the pipeline work in C using raspiraw.c as a starting point and could get the YUYV stream converted to I420 and displayed using the render component, however, I could not get the encoding to work. I kept ending up with a green output (basically just the headers I am assuming), so I think I am messing up the buffer handling. I am hoping that this python wrapper helps simplify that.

Thanks for the awesome work that you have done with raspiraw and for all your support!

6by9, 2018-04-11:
Headers won't give you green, however a buffer of all 0's will.

It depends on how you're passing the buffers from ISP to encode and render. 
I'm just working through the code for my [yavta fork](https://github.com/6by9/yavta/blob/master/yavta.c) again as that does exactly the pipeline you're after (though with V4L2 as the source). It pulls the buffers back to the ARM and uses mmal_buffer_header_replicate to create buffers that can be sent to two components. The replicated buffers hold a reference on the source buffer, and when both have been released it results in the source buffer being released so we can then send that back to the ISP for filling. A little involved, but is more efficient than using a video_splitter or similar which will copy the image data.

nsimon1, 2018-04-11:
@6by9 thank you for the additional advice and resources. I will let you know if I get it working, at least I know its definitely possible and I just need to learn how to work with the buffers properly. I appreciate your help with this!

closed 2018-04-17 by nsimon1

waveform80, 2018-09-29:
Note to self: read thru raspiraw.c again (might be several improvements to mmalobj, or the docs on the back of a better understanding)

waveform80, 2018-09-29:
Ah, excellent stuff - I'll bear that in mind when I'm going through mmalobj again! Thanks (and many thanks for jumping into numerous issues while I've been distracted with life and piwheels!)