waveform80/picamera #483: Unexpected zoom results

jcozar87, 2018-04-14:
As far as I understood from the documentation, `zoom` is composed of (x, y, w, h])

Let's say we have an image of `w, h = (3280, 2464)`. If I use a `zoom = (0.25, 0.25, 0.5, 0.5)` I expect to capture images where the center part matches with the center of the full image (like a x2 zoom). And the result is correct.

However, If I want the right part of the image (of resolution `w, h = (1640, 2464)`), I expect that using a `zoom=(0.5, 0, 0.5, 0.5)` gives me that part of the image. However, the following instruction modify the aspect ratio of the image and gives me an unexpected part of the image. What am I missing here?

```
camera = picamera.PiCamera(sensor_mode=2, resolution=(1640, 2464), framerate=10)
camera.zoom = (0.5, 0, 0.5, 1)
```

jcozar87, 2018-04-14:
I checked that using raspistill it is the same behaviour (forum link)[https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=210989&p=1302344#p1302344]. Therefore, I am missing some part of the documentation, or there is a bug somewhere.

jcozar87, 2018-04-14:
Ok oh @6by9 . Thank you for your explanation. I though it was just in the opposite order (first zoom and then re-scale to the requested resolution). Therefore, after the zoom is applied, I suppose that a re-scale to requested resolution is needed, and then is where aspect ratio can be different.

But in this way, I cannot achieve my objective: I see the full image coming off the sensor (full resolution 3280x2464), and now I want to select a rectangle of the image (my region of interest) and return such region with the original resolution.

As an example, let's say that the image is "splitted" into two pieces, the top and bottom. Let's say that my roi is the bottom half part (of resolution 1640x2464). How can I get this part of the image? (if it is possible...) 

Thank you!

jcozar87, 2018-04-16:
I have been playing a while with `sensor_mode`, `resolution`, `rotation` and `zoom` camera properties.

In this section of [picamera documentation](http://picamera.readthedocs.io/en/release-1.13/fov.html#division-of-labor), it explains how the library retrieves a frame. The third step ("Next the GPU’s image signal processor (ISP) performs several post-processing steps on the frame data.") descibes how the ISP process the raw image. The first one consists on "Transposition", which from the documentation I understand that it applies the `rotation`. The last process consists on resizing the raw image (until now, of resolution `sensor_mode`) to the requested `resolution`. However, the zoom property is not explained here. I suggest to add the zoom process here in the documentation (if I understood well, it is applied after the resizing step, and consists of applying the zoom to the ROI, and resize again to the requested resolution).

Finally, I suggest (if it is possible) to use the zoom property before the resizing step. At least for me, it has more sense to apply the zoom to the ROI based on the raw image (of `sensor_mode` resolution) and then resize the cropped image to the requested `resolution`.

6by9, 2018-04-16:
Documentation can be updated by anyone and submitted as a Pull Request.
Techincally resize and cropping (ie ROI) are done in a single stage at the end of the pipeline.

The behaviour of the firmware is not going to be changed as it would break all existing apps that make use of roi.
There is minimal saving within the GPU in having the strange crop done in the hardware pipe, so for most cases you may as well retrieve the full FOV and crop in the consumer if you want something particularly strange.

jcozar87, 2018-04-16:
Ok, thank you very much @6by9. I just wanted to clarify myself about how the pipeline of processes works. Now I fully understand the behaviour. I will make a pull request just in case you find it useful.

And of course, I understand that you cannot modify the actual behaviour. I also understand the rename of the property `crop` to `zoom` with respect to the previous version. Thanks!

closed 2018-04-16 by jcozar87