waveform80/picamera #486: Low quality images from mjpeg frame extraction

fedes82, 2018-04-18:
Hi Dave, I'm having some issues when trying to capture frames from a mjpeg stream. The thing is the images I get are really low quality jpgs, like this [https://imgur.com/a/YY1AS]()
Using the rapid capture and processing recipe as a guide mi program has to wait for a incoming tcp message to signal when to take the picture and then do a little processing in a separated thread. Images has to be 1920x1080 sized and the best quality possible. It gets aprox. 3 messages per second.
Beside of setting resolution, exposure_mode, exposure compensation, iso, shutter_speed, contrast, brightness and rotation, I use:

`camera.sensor_mode=2`
`camera.start_recording(output, format='mjpeg', quality=100)` 

When the main gets the signaling message, the `write` method of output searchs for the jpg magic number to start writing to a buffer, and stops writing to that buffer when it find the magic number again. If the main doesn't receive a signaling message, the `output`'s `write` method do nothing when called.

So, the questions would be, Can I get better images using frame extraction from mjpeg?, Given the output (3~4 1920x1080 images per second),  is `capture_continuos` a better choice? (on early stages of develpment I couldn't make my system work at that rates, so I changed it to the mjpeg way)
Thanks in advance.

6by9, 2018-04-19:
```quality``` is only relevant to image codecs (eg JPEG), not video codecs (eg MJPEG).
For video codecs you need to specify the ```bitrate```

6by9, 2018-04-21:
I'm working from knowledge of the firmware, not necessarily exactly how PiCamera is driving it.

Looking at the PiCamera code, quality is used at https://github.com/waveform80/picamera/blob/master/picamera/encoders.py#L790 as 
```
            self.output_port.params[mmal.MMAL_PARAMETER_VIDEO_ENCODE_INITIAL_QUANT] = quality
            self.output_port.params[mmal.MMAL_PARAMETER_VIDEO_ENCODE_MIN_QUANT] = quality
            self.output_port.params[mmal.MMAL_PARAMETER_VIDEO_ENCODE_MAX_QUANT] = quality
```
From memory those parameters have no meaning for MJPEG.

quality is used with the image_encode component to set the JPEG Q-Factor (1-100)

waveform80, 2018-09-28:
Ah, looks like I've at least got some documentation to update for this one!

waveform80, 2018-09-28:
(P.S. 6by9's answers are pretty much definitive - believe him over anything I've written in the docs!)

waveform80, 2018-11-03:
Hmm, just because it was trivial to do, I thought I'd just make absolutely sure the MJPEG codec was indeed ignoring the quality/quant value:

```python
for video in (False, True):
    for quality in range(1, 50):
        camera.capture('%s%02d.jpg' % (('still', 'video')[video], quality),
            use_video_port=video, quality=quality)
```

It's extremely rare that I ever contradict @6by9, but on this occasion I may have to as the results strongly suggest the MJPEG encoder is interpreting quality(/quant) in a similar manner to the JPEG image encoder:

```bash
$ ls -l *.jpg
-rw-r--r-- 1 dave dave  42437 Nov  3 22:48 still01.jpg
-rw-r--r-- 1 dave dave  43662 Nov  3 22:48 still02.jpg
-rw-r--r-- 1 dave dave  48667 Nov  3 22:48 still03.jpg
-rw-r--r-- 1 dave dave  54677 Nov  3 22:48 still04.jpg
-rw-r--r-- 1 dave dave  61562 Nov  3 22:48 still05.jpg
-rw-r--r-- 1 dave dave  68586 Nov  3 22:48 still06.jpg
-rw-r--r-- 1 dave dave  76607 Nov  3 22:48 still07.jpg
-rw-r--r-- 1 dave dave  83969 Nov  3 22:48 still08.jpg
-rw-r--r-- 1 dave dave  92963 Nov  3 22:48 still09.jpg
-rw-r--r-- 1 dave dave 100983 Nov  3 22:48 still10.jpg
-rw-r--r-- 1 dave dave 110663 Nov  3 22:48 still11.jpg
-rw-r--r-- 1 dave dave 121866 Nov  3 22:48 still12.jpg
-rw-r--r-- 1 dave dave 131537 Nov  3 22:48 still13.jpg
-rw-r--r-- 1 dave dave 140960 Nov  3 22:48 still14.jpg
-rw-r--r-- 1 dave dave 153694 Nov  3 22:48 still15.jpg
-rw-r--r-- 1 dave dave 164626 Nov  3 22:48 still16.jpg
-rw-r--r-- 1 dave dave 177834 Nov  3 22:48 still17.jpg
-rw-r--r-- 1 dave dave 187693 Nov  3 22:48 still18.jpg
-rw-r--r-- 1 dave dave 200162 Nov  3 22:48 still19.jpg
-rw-r--r-- 1 dave dave 209459 Nov  3 22:48 still20.jpg
-rw-r--r-- 1 dave dave 225267 Nov  3 22:48 still21.jpg
-rw-r--r-- 1 dave dave 240456 Nov  3 22:48 still22.jpg
-rw-r--r-- 1 dave dave 249685 Nov  3 22:48 still23.jpg
-rw-r--r-- 1 dave dave 265485 Nov  3 22:48 still24.jpg
-rw-r--r-- 1 dave dave 277868 Nov  3 22:48 still25.jpg
-rw-r--r-- 1 dave dave 287972 Nov  3 22:48 still26.jpg
-rw-r--r-- 1 dave dave 297020 Nov  3 22:48 still27.jpg
-rw-r--r-- 1 dave dave 304438 Nov  3 22:48 still28.jpg
-rw-r--r-- 1 dave dave 317840 Nov  3 22:48 still29.jpg
-rw-r--r-- 1 dave dave 331123 Nov  3 22:48 still30.jpg
-rw-r--r-- 1 dave dave 339459 Nov  3 22:48 still31.jpg
-rw-r--r-- 1 dave dave 347687 Nov  3 22:48 still32.jpg
-rw-r--r-- 1 dave dave 365970 Nov  3 22:48 still33.jpg
-rw-r--r-- 1 dave dave 366981 Nov  3 22:48 still34.jpg
-rw-r--r-- 1 dave dave 376349 Nov  3 22:48 still35.jpg
-rw-r--r-- 1 dave dave 397562 Nov  3 22:48 still36.jpg
-rw-r--r-- 1 dave dave 397577 Nov  3 22:48 still37.jpg
-rw-r--r-- 1 dave dave 408317 Nov  3 22:48 still38.jpg
-rw-r--r-- 1 dave dave 408667 Nov  3 22:48 still39.jpg
-rw-r--r-- 1 dave dave 407662 Nov  3 22:48 still40.jpg
-rw-r--r-- 1 dave dave 421306 Nov  3 22:48 still41.jpg
-rw-r--r-- 1 dave dave 420302 Nov  3 22:48 still42.jpg
-rw-r--r-- 1 dave dave 471010 Nov  3 22:48 still43.jpg
-rw-r--r-- 1 dave dave 470737 Nov  3 22:48 still44.jpg
-rw-r--r-- 1 dave dave 472135 Nov  3 22:48 still45.jpg
-rw-r--r-- 1 dave dave 471633 Nov  3 22:48 still46.jpg
-rw-r--r-- 1 dave dave 486567 Nov  3 22:48 still47.jpg
-rw-r--r-- 1 dave dave 487882 Nov  3 22:48 still48.jpg
-rw-r--r-- 1 dave dave 487367 Nov  3 22:48 still49.jpg
-rw-r--r-- 1 dave dave  17380 Nov  3 22:48 video01.jpg
-rw-r--r-- 1 dave dave  18537 Nov  3 22:48 video02.jpg
-rw-r--r-- 1 dave dave  22783 Nov  3 22:48 video03.jpg
-rw-r--r-- 1 dave dave  27847 Nov  3 22:48 video04.jpg
-rw-r--r-- 1 dave dave  33553 Nov  3 22:48 video05.jpg
-rw-r--r-- 1 dave dave  39350 Nov  3 22:48 video06.jpg
-rw-r--r-- 1 dave dave  46439 Nov  3 22:48 video07.jpg
-rw-r--r-- 1 dave dave  53453 Nov  3 22:48 video08.jpg
[... 19 lines cut ...]
-rw-r--r-- 1 dave dave 256198 Nov  3 22:48 video28.jpg
-rw-r--r-- 1 dave dave 268696 Nov  3 22:48 video29.jpg
-rw-r--r-- 1 dave dave 277118 Nov  3 22:48 video30.jpg
-rw-r--r-- 1 dave dave 285801 Nov  3 22:49 video31.jpg
-rw-r--r-- 1 dave dave 290179 Nov  3 22:49 video32.jpg
-rw-r--r-- 1 dave dave 302496 Nov  3 22:49 video33.jpg
-rw-r--r-- 1 dave dave 302582 Nov  3 22:49 video34.jpg
-rw-r--r-- 1 dave dave 311712 Nov  3 22:49 video35.jpg
-rw-r--r-- 1 dave dave 324449 Nov  3 22:49 video36.jpg
-rw-r--r-- 1 dave dave 324290 Nov  3 22:49 video37.jpg
-rw-r--r-- 1 dave dave 333306 Nov  3 22:49 video38.jpg
-rw-r--r-- 1 dave dave 332670 Nov  3 22:49 video39.jpg
-rw-r--r-- 1 dave dave 333332 Nov  3 22:49 video40.jpg
-rw-r--r-- 1 dave dave 344965 Nov  3 22:49 video41.jpg
-rw-r--r-- 1 dave dave 344625 Nov  3 22:49 video42.jpg
-rw-r--r-- 1 dave dave 379887 Nov  3 22:49 video43.jpg
-rw-r--r-- 1 dave dave 379278 Nov  3 22:49 video44.jpg
-rw-r--r-- 1 dave dave 379820 Nov  3 22:49 video45.jpg
-rw-r--r-- 1 dave dave 379363 Nov  3 22:49 video46.jpg
-rw-r--r-- 1 dave dave 394938 Nov  3 22:49 video47.jpg
-rw-r--r-- 1 dave dave 395111 Nov  3 22:49 video48.jpg
-rw-r--r-- 1 dave dave 395591 Nov  3 22:49 video49.jpg
```

Incidentally, just to address @fedes82 original query: if your issue is the quality of the frame from the video encoder, different capture methods will make no difference. Recording video, capture(_continuous, etc.) with use_video_port=True - they all drive the same encoding pipeline (and video encoder) under the covers so the quality will be the same.

6by9, 2018-11-05:
> Hmm, just because it was trivial to do, I thought I'd just make absolutely sure the MJPEG codec was indeed ignoring the quality/quant value:

```
for video in (False, True):
    for quality in range(1, 50):
        camera.capture('%s%02d.jpg' % (('still', 'video')[video], quality),
            use_video_port=video, quality=quality)
```

Isn't ```camera.capture``` using image_encode and hence the JPEG codec, not video_encode and the MJPEG codec? It's the encoding component that matters, not which camera component port is in use.

waveform80, 2018-11-09:
> Isn't camera.capture using image_encode and hence the JPEG codec, not video_encode and the MJPEG codec? It's the encoding component that matters, not which camera component port is in use.

Doh! Absolutely correct. It's connecting `image_encode` to the splitter port (which is connected to the video port). I should've shoved the DEBUG bit on and checked the pipeline! Well, some doc fixes necessary after all (once my head's cleared this damned cold)

commit 874ed339 (2018-11-15): Fix #486

Quality is ignored for the MJPEG video encoder (but not the JPEG
image encoder)
closed 2018-11-15 by waveform80