waveform80/picamera #488: picamera: failed to enable connection: Out of resources

Daabjeerin, 2018-05-09:
>python main.py
mmal: mmal_vc_port_enable: failed to enable port vc.null_sink:in:0(OPQV): ENOSPC
mmal: mmal_port_enable: failed to enable connected port (vc.null_sink:in:0(OPQV))0xef54c0 (ENOSPC)
mmal: mmal_connection_enable: output port couldn't be enabled
Traceback (most recent call last):
  File "main.py", line 10, in <module>
    video_camera = VideoCamera(flip=True) # creates a camera object, flip vertically
  File "/home/pi/Smart-Security-Camera/camera.py", line 9, in __init__
    self.vs = PiVideoStream().start()
  File "/usr/local/lib/python2.7/dist-packages/imutils/video/pivideostream.py", line 10, in __init__
    self.camera = PiCamera()
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 433, in __init__
    self._init_preview()
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 513, in _init_preview
    self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
  File "/usr/lib/python2.7/dist-packages/picamera/renderers.py", line 558, in __init__
    self.renderer.inputs[0].connect(source).enable()
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 2212, in enable
    prefix="Failed to enable connection")
  File "/usr/lib/python2.7/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources
Exception AttributeError: "'VideoCamera' object has no attribute 'vs'" in <bound method VideoCamera.__del__ of <camera.VideoCamera object at 0x6e41b350>> ignored

granpino, 2018-05-23:
Out of resources means the camera is being used by another app.

jazzlorv, 2019-02-05:
I keep getting this error. I recently tried a lot of methods in using my python script to run at pi startup. After few hours of successful runs it just giving me this error.

adriandegonda, 2019-05-10:
> Out of resources means the camera is being used by another app.

I got the exactly same error, then  restarted PyCharm and my Raspberry Pi and it worked afterwards. 
Thanks alot!

SdeGeata, 2019-06-26:
How does one restart PyCharm using terminal?

pmne324, 2020-08-02:
I have same problem but I don't now how to fix it. Please help!
Thanks all.

yoganand81, 2020-08-16:
I faced this problem recently. I was using picamera to click a pic and then send to HDMI display using fbi. The program will run twice before throwing this error.
After I killed the running fbi processes, the issue got resolved!

DmitryRomantsov, 2021-03-22:
The script runs copies of itself.

abdullahmujahidali, 2021-05-20:
how to kill running fbi processes?

ruitcatarino, 2022-09-13:
Adding a camera.close() after camera.stop_preview() solved the issue for me.

ufoozhenghao, 2023-05-11:
anyone can help? i get the same issue but i‘m sure there isn't any other app using camera

my code:
```
from time import sleep
from picamera import PiCamera

camera = PiCamera()
#camera.resolution = (1024, 768)
camera.start_preview()
# Camera warm-up time
sleep(2)
camera.capture('foo.jpg')
camera.stop_preview()
camera.close()
```

error:
[image]