waveform80/picamera #489: Capturing timelapse sequences

mfraser, 2018-05-10:
Tried to run this example:
import time
import picamera
from datetime import datetime, timedelta

def wait():
    # Calculate the delay to the start of the next hour
    next_hour = (datetime.now() + timedelta(hour=1)).replace(
        minute=0, second=0, microsecond=0)
    delay = (next_hour - datetime.now()).seconds
    time.sleep(delay)

with picamera.PiCamera() as camera:
    camera.start_preview()
    wait()
    for filename in camera.capture_continuous('img{timestamp:%Y-%m-%d-%H-%M}.jpg'):
        print('Captured %s' % filename)
        wait()

but was getting:
TypeError: 'hour' is an invalid keyword argument for this function

Changing hour to hours and it works.

AdamMajer, 2018-07-01:
`hours=1` not `hour`

waveform80, 2018-11-03:
Doh again! Already fixed in 6817fbc2fc667924488897272f35e2ddc96c13fb

closed 2018-11-03 by waveform80