waveform80/picamera #491: Stream h264 to browser

Esser420, 2018-05-21:
First of all let me thank you for this incredibly powerful library!

I'm working on a video streaming project with raspberries and found the streaming to web recipe very useful and easy to use.

However I'm running 4 raspberries (soon to be more) streaming to a streaming server. The problem I have is with the bandwidth, it is way to large, about 16Mbit/s per stream.. This makes watching multiple streams at once kind of difficult and also puts a lot of stress on the network.

This is naturally due to the video format (mjpeg), the individual jpegs are just too big (around 80Kb).

So I was looking at actually streaming in h264 format, which would reduce the bandwidth significantly!

The problem is I have not found any way to display h264 streams in a browser.

Is there a moderately simple way to achieve this? Are there any other projects I can take a look at to get started? Basically I would like something running on a browser that simply reads the stream produced by ```start_recording(socket, format = "h264")```

vedmant, 2018-07-15:
Hi, I'm actually working on this too, but I used node.js approach, I use node.js package https://github.com/pimterry/raspivid-stream , working demo https://github.com/pimterry/pi-cam

Esser420, 2018-07-18:
@vedmant That's pretty cool!

I did eventually figure it out, used the same javascript decoding library as you. I did however run into some issues. For example multiple streams (from multiple cameras) on the same page. The http-live-player.js is not really prepared to be used as a worker, which is something the Broadway decoder was actually prepared for. So I ended up adapting this one: https://github.com/Dregu/visio. So now I am able to have 1 worker per stream which means that multiple streams on same page will run the same speed.

Another issue I had was that some browsers (specially mobile) do not work very well with the library. So for this special use case I ended up using the streaming example from @waveform80: https://github.com/waveform80/pistreaming

Mobile browsers are able to handle it nicely.

vedmant, 2018-07-18:
@Esser420 That's great, I'll check that out. I also want to find a way to detect motion from the h264 steam using nodejs, found only python and c++ code samples, but nothing that really works out of the box.

waveform80, 2018-09-28:
Intriguing; ages ago I wrote [pistreaming project](https://github.com/waveform80/pistreaming.git) to demo web-streaming but concluded (back then) that H.264 was a non-starter due to lack of wide support (see extended rant at the end of the pistreaming README ;-). That's largely been corrected in the years since, but (not being a heavy Javascript user) I haven't kept up with what's been happening on the client side. Looks like pistreaming could probably do with a re-visit at some point if there's better support these days.

Anyway - I'll close this for now as I don't think I can add anything more at this stage, but do feel free to re-open it if you've further questions!

closed 2018-09-28 by waveform80