waveform80/picamera #493: Overlay stopped workign after update to PiCamera 1.13

naturephotographer, 2018-06-03:
I was testing out the use of overlays and got the example from readthedocs on overlays to work in Python 3 using PiCamera 1.12 after replacing np.getbuffer(a) with memoryview(a) but just after updating to PiCamera 1.13 this same code:
o = camera.add_overlay(memoryview(a), layer=3, alpha=64)
now gets this error:
...
File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 466, in buffer_bytes
return m.itemsize * reduce(mul, m.shape)
UnboundLocalError: local variable 'm' referenced before assignment

When I check this routine in gethub I see that 'm' does appear to be undeclared if the if statement is false.  How can this be fixed?

def buffer_bytes(buf):
    """
    Given an object which implements the :ref:`buffer protocol
    <bufferobjects>`, this function returns the size of the object in bytes.
    The object can be multi-dimensional or include items larger than byte-size.
    """
    if not isinstance(buf, memoryview):
        m = memoryview(buf)
return m.itemsize * reduce(mul, m.shape)

naturephotographer, 2018-06-03:
The cut and paste from gethub did not preserve the indenture. here is the corrected version assuming this preserves the indenture when I save.

def buffer_bytes(buf):
    """
    Given an object which implements the :ref:`buffer protocol
    <bufferobjects>`, this function returns the size of the object in bytes.
    The object can be multi-dimensional or include items larger than byte-size.
    """
    if not isinstance(buf, memoryview):
        m = memoryview(buf)
return m.itemsize * reduce(mul, m.shape)

Note that 'm' is not defined when the if statement is false.

naturephotographer, 2018-06-03:
The indenture was correct prior to saving but the save removed it so to see the original see:
https://github.com/waveform80/picamera/blob/master/picamera/mmalobj.py
starting on line 458

michaelkevinjackson, 2018-06-11:
I found similar error report when trying the example code from http://picamera.readthedocs.io/en/release-1.13/recipes2.html#unencoded-video-capture

I think there may just be a bug in the statement at line 464, which is
    if not isinstance(buf, memoryview):
in release 1.13. If I remove the "not" and put this instead:
    if isinstance(buf, memoryview):
it seems to work fine. It was failing when the condition isinstance(buf, memoryview) was true.

naturephotographer, 2018-06-11:
Thank you for the response.

This is my first use of Python and a Raspberry Pi. 
Is accomplishing this as simple as finding the mmalobj.py module and deleting the "not" and it will work? I was reading on Modules and apparently even if there is a compiled version, Python will check to see if there is an update and recompile at start up when running my program.
I discovered that this exact problem was reported here a year ago and was never fixed so I wonder why PiCamera continues to be included in Raspbian with this error uncorrected especially if it is so simple to fix? see https://github.com/waveform80/picamera/issues/419

naturephotographer, 2018-06-11:
Not being familiar with Raspbian or Python I had to do a little research to find I had to use the terminal to gain superuser access to be able to save the modified mmalobj.py and found that nano came with Raspbian Jessie and was user friendly, not requiring learning commands for a new line editor. I also had to first find the Python 3 version of mmalobj.py to edit. This was easy with a search for picamera and selecting the one that was under Python3.
Once I removed the "not" on line 464 this now works so whomever is responsible for PiCamera 1.13 needs to fix this so more people do not have to go through all this trouble and not have to hack the system files.
Thank You.

6by9, 2018-06-13:
@michaelkevinjackson @naturephotographer If you feel able to, then fork the project and create a pull request with the fix. That's how open source coding works.

See https://picamera.readthedocs.io/en/latest/development.html#development-installation for basic instructions on downloading a development environment.
Or for such a simple change you could go in with the web editor (https://github.com/waveform80/picamera/blob/master/picamera/mmalobj.py and click on the pencil icon at the top right) to make the edit.
Once you've committed the change to your fork of the project, go to the Pull Requests tab of your forked project and hit "New Pull Request". Follow the prompts and it should create a change that can be reviewed.

wilcovantoorn, 2018-08-31:
I guess the pull request wasnt done, so i did a pull request with the bug fix

waveform80, 2018-09-28:
Argh, definitely a bug - will get this fixed for 1.14. I'm afraid removing "not" isn't actually the correct fix; replacing "m" with "buf" in the following line is (the intent is that if buf isn't already a memoryview it should be converted to one, and the subsequent line should then deal with the definitely-a-memoryview-buf).

waveform80, 2018-09-28:
P.S. many thanks to @wilcovantoorn for the PR - it may not be strictly correct, but I greatly appreciate it when people take the time to make a PR as it always make my life easier even when it needs work!

naturephotographer, 2018-09-29:
I realized the "not" must have been intended and if it was removed it might cause problems elsewhere which is why I thought it was a bad idea to do a pull request with this modification of just removing the "not" and why I reported the issue back in June. I didn't want to just fix my problem and cause problems for other people.

I was the one that tracked the issue to that area of code but was not sure about modifying released code until michaelkevinjackson tested removing the "not".

I realized that whomever is familiar with this software should be the one(s) to see what might be wrong as I did not want to try and locate all the places this gets used and the potential impacts.

In my issue reporting I also mentioned that the exact same problem was reported a full year earlier in June of 2017 (issue #419) but apparently never addressed.
Thank you for looking at this and explaining the correct solution.
Don

wilcovantoorn, 2018-09-29:
Youre welcome. We always try to help everybody as good as we can. But thank
you dave. I appreciate it! We are at the makerfaire in eindhoven holland at
the moment and it is great to meet so many people.

Gr. Wilco van Toorn

waveform80, 2018-11-03:
Fixed in 96b8a0983194dfa7e7d2fd19ab4317961ba1ea44

closed 2018-11-03 by waveform80

felric, 2018-12-20:
I recently upgraded distro to Stretch.
After doing all updates, I'm not able to see the preview on HDMI screen. I tried both with python PiCamera and raspistill -v, both in fullscreen and window mode. I tried to set up graphics memory to 256 Mb but it did not help. 

However, even if preview is not working at all, taking pictures and creating JPG works flawlessly.

To sum up: no way to get preview on HDMI output after update to Stretch and PiCamera v.1.13, but it outputs snapshots to JPG.
Have you any hint to solve this issue?

Thanks
Riccardo