waveform80/picamera #495: Disabling AGC, AWB and friends

Krakawet, 2018-07-02:
Hello !

I'm trying to get the more stable picamera image parameters as possible, in order to do cools things like colorimetry, that require constant sensitivity to light and color.

In order to achieve this, I installed the last commits and set these settings to a fixed value : 
Capture parameters :
- Shutter speed : 11111

Automatic Gain control :
- exposure_mode : off
- analog_gain: 8.0
- digital_gain: 1.2

Auto-White Balance  : 
- awb_mode : off

Exposure time, awb and acg gains are now proudly fixed, but it seems I still have some color adaption algorithm running somewhere: 
1 ) My white background looks snow-white.
2 ) I present a finger in front of the camera.
3 ) The background goes a bit redish (transition take around 1 second)

Did I missed to fix some setting somewhere ? 
Is it just possible to achieve my goal ?

Personal note : Thank you 6by9, rwb27 and Dave, you are my everyday heroes !

yalcinbar, 2018-09-16:
Setting "awb" and "agc" to "off" is simply not enough for your case. You also need to get rid of the color processing blocks in ISP. Here are the things you can do:
1 - [Capturing RAW images](https://picamera.readthedocs.io/en/release-1.13/recipes2.html#raw-bayer-data-captures): RAW image data come directly out of the sensor, without any processing applied. Since you are interested in the response of the sensor to the incoming light, this is the ideal solution for your case. But, keep in mind that you may need to do certain post-processing (like lens-shading correction) to compensate for the nonuniformities. 
2 - [Disabling ISP blocks](https://www.raspberrypi.org/forums/viewtopic.php?t=175711): This is the tedious trial and error approach. Just disable the ISP blocks one by one, until you get the desired output.

waveform80, 2018-09-28:
Disabling ISP blocks is something I need to add (as an advanced feature) in 1.14; not sure if I made a ticket for this already so this'll do to remind me for now!

6by9, 2018-11-05:
FYI I'm adding an extra controllable bit to MMAL_PARAMETER_CAMERA_ISP_BLOCK_OVERRIDE that allows you to disable demosaicing as well. Bit 11.
It makes little sense with a Bayer sensor (you get an odd checkerboard effect), but some people may have managed to obtain mono versions of these sensors.

waveform80, 2018-11-15:
Implemented in 949421b231e0a0396e8dff2a348824e5fa5006ee

closed 2018-11-15 by waveform80