waveform80/picamera #498: Search on the documentation website doesn't work properly

soswow, 2018-07-22:
Search for "opencv" for example - https://picamera.readthedocs.io/en/release-1.13/search.html?q=opencv&check_keywords=yes&area=default
I get two results. The second result lead to https://picamera.readthedocs.io/en/release-1.13/recipes1.rst.html?highlight=opencv which is 404. The correct would be https://picamera.readthedocs.io/en/release-1.13/recipes1.html?highlight=opencv

waveform80, 2018-09-28:
Huh ... that's bizarre. I'm not sure whether this is an issue in picamera's docs config or an issue on readthedocs but I'll have a dig into it for 1.14!

waveform80, 2018-11-03:
This appears to have been an upstream issue rtfd/readthedocs.org#2914 which is now resolved. I've regenerated the 1.13 docs and the search appears to work (be aware you may need to clear your browser cache to notice the fix); it's also working on the "latest" docs version which'll eventually become the 1.14 docs.

closed 2018-11-03 by waveform80