waveform80/picamera #500: Wrong timing in videosimport picamera  camera = picamera.PiCamera() camera.resolution = (640, 480) camera.start_recording('/Photos/tmp/my_video.h264') camera.wait_recording(30) camera.stop_recording()

Krakawet, 2018-08-16:


6by9, 2018-08-16:
An H264 elementary stream has no timestamp or frame rate information. When you run ffmpeg of MP4Box it is guessing at the frame rate, and normally guesses 25fps. 30s * 30/25 = 36secs.

Tell ffmpeg the actual framerate via the -r flag (```ffmpeg -r 30 -i test.h264 -c:v copy -f avi test.avi``` it matters where it goes), or the equivalent with MP4Box, and you should get an output file playing back at the same speed.

Krakawet, 2018-08-20:
Thank you6by9, it's almost working !
Running on this bare minimal exemple, framerate is now correct.

Unfortunetly,  it's still slow when capturing/encoding videos in my real and massive program, but I managed to get a correct framerate using, 
`ffmpeg -r 40 -i my_video.h264 -c:v copy -f avi my_video.avi`, despite the fact that the camera framerate was (30,1) while recording...

I'm working on a tentaculous project with multiple commits (noaspect and lens shading correction) and a lot of other specificities, but can do some test if it helps the community !
Thank you !

waveform80, 2018-09-28:
Surely "-r 40" (framerate 40) implies that the camera was too fast? Anyway, I can't reproduce this with a minimal example here. If you can whittle your tentaculous project down to a minimal reproduction case, I can take a further look - do feel free to re-open this issue in that case!

closed 2018-09-28 by waveform80