waveform80/picamera #504: OSError: libbcm_host: cannot open shared object file: No such file or directory.

yujmo, 2018-09-09:
RancherOS Version: (ros os version)
v1.4.0
Where are you running RancherOS? (docker-machine, AWS, GCE, baremetal, etc.)
Raspberry Pi

When I try to use image[ubuntu:armhf] to control sensors in rancheros, I can use GPIO to control the sensor.

However, when I use the python module: picamera to control the Human body induction sensor sensor, it is wrong.The wrong message is: OSError: libbcm_host: cannot open shared object file: No such file or directory.

I have downloaded libbcm_host.so from GitHub and copied it to /usr/local/lib and /usr/lib, but still report the same error. I need help. Thanks.🙂🙂🙂

6by9, 2018-09-10:
Are you going for the scatter-gun effect of opening this issue everywhere? https://github.com/raspberrypi/firmware/issues/1034
https://github.com/rancher/os/issues/2472

- Test the Ubuntu image natively before introducing RancherOS into the mix.
- Download https://github.com/raspberrypi/userland/ (which is where the bcm_host source is) and build it for yourself (run the ./buildme script).

waveform80, 2018-09-28:
> However, when I use the python module: picamera to control the Human body induction sensor

The *what*? I didn't know my library could control that ;)

Anyway, definitely not a picamera issue!

closed 2018-09-28 by waveform80

MaksimDan, 2020-05-18:
Leaving this as a note if anyone else gets here, followed the recommend advice to

`./buildme --aarch64`

then added 
`sudo cp libbcm_host.so /opt/vc/lib/`

and reran but got
`OSError: libbcm_host.so: wrong ELF class: ELFCLASS32`

So it seems that libbcm_host.so are one of those files where 64-bit userspace is not officially supported for.