waveform80/picamera #505: Is it possible to preview only a part of the image (basically roi)

Christanoid, 2018-09-09:
Hello,
first up, sorry for not marking this as a question, I am new to github and don't know how to do this, so it will remain as a "issue".
Anyhow, now to the question I have. I am currently building myself a little home cinema featuring philips hue and controlling the lamps using huestacean. Due to my video source coming from inside my smart TV I can't extract the image from the TV to huestacean, so I came up with the probably stupidest idea to surpass that problem. Use an old PS3 Eye I have laying around, preview the recording from the webcam on a Raspberry pi 3 and let Huestacean film that webcam preview.
My problem now is, that I can't put the optical zoom of the PS3 eye to exactly match the TV and therefore need to cut away everything around the TV. (Huestacean uses the edges of the monitor it sees so I have to remove my shelves out of the image and scale the TV part to be all of the screen). I can use hardcoded area borders seeing how I won't move the camera nor the TV, but I don't know how to cut party away or if I can even do that with this library.
Additionally my camera is also slightly below the TV because I want to hide it under the Bed/Couch, so is it also possible to morph the image making the bottom a bit wider?

Thanks for the help,
Chris

6by9, 2018-09-10:
This repo is a Python library for the Raspberry Pi cameras only (it uses the hardware blocks of the Broadcom SoC).
The PS3 Eye is a USB webcam, therefore not relevant here. You need to be looking at the V4L2 APIs. You're probably best off asking on the forums (https://www.raspberrypi.org/forums/index.php) rather than Github, however whilst you may be able to switch between the two optical zoom modes (is that even software controlled?), you are very unlikely to be able to crop the image as you want.
You can see if the PS3 Eye supports the selection API (https://linuxtv.org/downloads/v4l-dvb-apis-new/uapi/v4l/selection-api-intro.html), but I haven't encountered a webcam as yet that does.

Christanoid, 2018-09-11:
Ah ok, thought I read somewhere, that this library can also be used on USB cameras, but I guess not. Thanks for the hint regarding the V4L2 library, I'll read into that in the next days, maybe I can find a method to do what I need there.
Oh and btw, the two zoom options of the PS3 Eye are controlled by physically turning a part of the Eye, so no software control. Unfortunately even the zoomed mode doesn't go quite far enough, so I'll still have to find a way to cut a part away, maybe that selection API does, I'll see within the next few days when I try to work my way into the other library.

closed 2018-09-28 by waveform80