waveform80/picamera #506: Add max available resolution and /or modes list as info property to camera class

Aydan, 2018-09-15:
I was just trying to programmatically find out the maximum resolution of the attached camera, but I didn't find any property with this information.
So I'd like to suggest an extra property like native_resolution or some such.
Also a list of available modes and their resolution/binning values would be nice so one could programmatically select the correct mode.

yalcinbar, 2018-09-17:
1- Maximum resolution depends on the camera module and GPU's capabilities. You can get it by :
```
from picamera import PiCamera
# this returns the MAX_RESOLUTION tuple
max_res = PiCamera.MAX_RESOLUTION
```
2 - Sensor modes are listed [here](https://picamera.readthedocs.io/en/release-1.13/fov.html#sensor-modes).

Aydan, 2018-09-17:
@yalcinbar: I must have overlooked the "constant" for max resolution. I was looking for an attribute. My bad.
It is also not mentioned anywhere in the camera class api, only in one FAQ about preview flickers with the V2 camera, and of course the source code. Maybe this could be added to the documentation of the camera class.

for the sensor modes: I did find the mode tables, but it might be useful to have them accessible programmatically via the camera api.

waveform80, 2018-09-29:
Hmmm, that's a good suggestion. Probably needs similar logic to PiCamera.MAX_RESOLUTION (changes according to attached hardware) but I like the idea of a simple dict containing sensor modes mapped to resolutions and framerates (and perhaps other details).

closed 2018-11-04 by waveform80

Aydan, 2018-11-05:
In my opinion the resolution in the sensor mode should be a tuple containing width and height, not a string.

commit 51179149 (2018-11-04): Closes #506

Added sensor_modes property to list the available sensor modes for the
attached camera.