waveform80/picamera #507: What does the library need to run in a Docker Container?

Esser420, 2018-09-24:
I'm trying to build a service that really needs this library.

The only way I found to run this inside a ocntianer was with the --priviledged flag.

The problem with that is that I cannot run it as a service since Docker doesn't allow "privileged" services. 

Are there any other tricks I could use? like mounting some directory inside it?

waveform80, 2018-09-29:
The camera needs access to the /dev/vchiq device at the least. I'm afraid I can't add much more than that (my Docker knowledge is minimal), but the camera is very much a "privileged" device in the sense that only a single process can access it at a time, and there's plenty of special handling (reserved memory for the GPU, etc.) that surrounds it (read the [camera hardware chapter](https://picamera.readthedocs.io/en/release-1.13/fov.html) for a view of just how "special" :), hence I'd be rather surprised if this could be made to run in unprivileged containers.

Esser50K, 2021-05-25:
Issue solved, to get it running inside a docker container one needs to add the following flags to the `docker run` command:
`--privileged -v /opt/vc:/opt/vc -e LD_LIBRARY_PATH=/opt/vc/lib`

closed 2021-05-25 by Esser50K