waveform80/picamera #509: Quick capture and rendering on a widget

c30ra, 2018-10-01:
Can some one point me to an example, or give me a snippet on how to capture image from raspicam? Basically I need to do the same thing raspivid does, same latency of rendering. But it must be rendered on a Qwidget. What I'm doing now is using mmal object and use the call back function to grab the frame. But it's still too slow

waveform80, 2018-10-02:
raspivid uses the camera's preview system to achieve its low latency. As [FAQ #2](https://picamera.readthedocs.io/en/release-1.13/faq.html#can-i-put-the-preview-in-a-window) in the docs explains, this essentially precludes interfacing with a windowing system.

I should change the wording in that answer really - the preview system isn't "crude". Rather, it has outputs that make perfect sense in the mobile world (where the preview should fill the screen and have as little latency as possible, so don't even bother copying the frames from GPU to CPU; just deal with them immediately in the GPU). Unfortunately, in the desktop world those outputs are slightly at odds with a windowing system that wants to place things on-screen too.

Anyway, I'd suggest there's little chance of coming up with a preview system that places frames within a windowed widget with the same latency as raspivid's preview (simply because it will necessarily entail getting frame data to the windowing system for it to position). But lower your expectations slightly, and it's possible to come up with a tolerable preview system by rendering constant captures within a window.

c30ra, 2018-10-02:
Actually I'm come up with this:

```
from picamera import mmal, mmalobj as mo
from time import sleep

def image_callback(port, buf):
    print(buf.data) # in my real code i store this in a QImage to use it in a QWidget
    return False # this should tell the camera to continue to stream, right?

camera = mo.MMALCamera()

camera.outputs[0].framesize = (1920, 1080)
camera.outputs[0].framerate = 30
camera.outputs[0].format = mmal.MMAL_ENCODING_RGB24
camera.outputs[0].commit()

camera.outputs[0].enable(image_callback)

sleep (10)

camera.outputs[0].disable()
```

this is a Snippet, it's not the actual code. However, if I have understood the documentation correctly, in this code I'm using the preview channel of the camera. So this is the best I can get? With this I have 40+ ms of latency, that is acceptable, but was wondering if it 's the correct way to do it, or there are fast alternatives... But I think there aren't if I Understand you answer correctly

waveform80, 2018-10-03:
Yup, that's using the preview channel (0). It's not the absolute best you can get because picamera doesn't fully utilize the zero-copy mode of MMAL, i.e. picamera's mmalobj layer always copies buffers before handing them to callbacks so that Python owns the buffer (purely a usability optimization).

If it's compatible with QWidget (I've no idea), you might be able to get lower by using C with MMAL's zero-copy buffers, but that would entail having a QImage that doesn't own the buffer and ensuring that you know when it's actually rendered so you can hand the buffer back in time. That would probably shave a fair bit off as 1080p frames aren't small and copying them around is a considerable cost.

However, you need to drop your expectation of the "same latency of rendering" as raspivid. That's using the built-in preview system in which the CPU isn't even involved in requesting frames be drawn; it's the GPU compositing frames from the camera onto the display. It's simply unrealistic to expect to get *that* low.

c30ra, 2018-10-04:
Unfortunately, Python is the only way since the project I'm working on doesn't need all that speed. Plus it saves me from the "write, compile, debugs" loop. It's just an interesting/research purpose if would be possible to achieve that frame rate. for the QWidget, I'm using the QImage as you said and render the image directly. Anyway thanks for the answer!!!

closed 2018-10-04 by c30ra

c30ra, 2018-10-05:
sorry, but i'm not an expert on camera acquisition or hardware in general. So I'm using both terms in the same way, what was interested in is latency, when I see something moves, I would like to see it in real time on my computer screen. Actually, I haven't measured the fps, I'm measuring the time it takes the thread.wait() to tell me the frame is ready (not sure if it the best way to measure it). From my tests, I get 35ms to 54ms from a frame to another. The widget instead as an update rate set to 16ms. so every time it enters the paintEvent() it wait for the frame if it'not ready. But since my update rate is faster than the frame grabbing I always wait for it. So the FPS should vary on how fast a frame is returned from the callback. Since I have an even 40ms of wait, it should be around 25 FPS.