waveform80/picamera #511: Minimum shutter with Auto ISO

GuidoBartoli, 2018-10-03:
Hi,
I'm currently using picamera in a project involving a camera mounted on a robot facing the ground in front of it. For image classification purposes, I need this system to have a shutter speed shorter than 1/250 s (4000 us of exposure time) to avoid motion blur and ISO should be set consequently. So I tried to setup camera with these simple settings (maybe `sport mode` is not necessary?):
```
exposure_mode = 'sports'
shutter_speed = 4000
iso = 0
```
This works fine in most situation (ISO is dynamically changed to maintain an average exposure with constant shutter speed), but since this robot is designed to work outdoors, sometimes the sun is really strong on the ground and a faster shutter speed is needed to avoid over-exposure and a washed-out image impossible to analyze.

It would be useful to set also a _minimum_ shutter speed the works in conjunction with _automatic_ ISO and exposure, so setting `shutter_speed = 0`, `iso = 0` and `min_shutter = 4000` would allow the algorithm to behave like this (pseudo code):
1. Start with minimum ISO and shutter speed computed to have a balanced exposure
1. Continuously check exposure level
    1. If exposure is still balanced, leave current settings untouched
    1. Else if over-exposure is detected, reduce shutter speed maintaining the same ISO
    1. Else if under-exposure is detected, increase shutter speed maintaining the same ISO
        1. If shutter speed is under the minimum value, progressively increase ISO 
        1. Else if shutter speed is minimum, ISO at maximum and under-exposure is still present, nothing can be done
1. Repeat from step 2.

In modern digital cameras, this is commonly called "aperture priority with auto ISO and minimum shutter speed", and in some models the latter is automatically chosen by the camera accordingly to the lens focal length to avoid shaking hands blur. Often the maximum allowed ISO value can also be chosen.

Do you think this can be added to already wonderful picamera library?

Regards?

waveform80, 2018-10-03:
picamera is a (formerly thin, now slightly chubby) layer over the camera's firmware so basically the answer is "if the firmware implements it, then sooner or later (sometimes much later) picamera will probably expose it". At the moment, I'm not aware of such a facility in the firmware, but I'm still ploughing through a year's backlog of tickets, PRs, forum posts, and header diffs to catch up on stuff and there's plenty of new things I wasn't aware of.

So ... I don't *think* so, but if anyone wants to point out a new bit in MMAL I've missed that implements this, feel free?

6by9, 2018-10-04:
Not supported by the firmware, and not something I would like to try and plumb in. Sorry.