waveform80/picamera #512: Frames darker with formats `'h264'` and `'rgb'` than formats `'mjpeg'` and `'yuv'`

sanntann, 2018-10-04:
Hi,

I'm using `picamera` to capture and process video in low light conditions. It seems to me that some formats provide better contrast in such conditions than others. Specifically, `'mjpeg'` and `'yuv'` formats greatly outperform `'h264'` and `'rgb'` formats. Unfortunately, I am limited to using `'h264'` for video capture, as no other format can be written to disk at 30 FPS and with resolution 1600 x 1216.

Is there a way to achieve image quality with `'h264'` that is comparable to `'mjpeg'` and `'yuv'`?

I have tried with different values for `'h264'` additional options, such as _profile_, _level_, _intra_period_, _bitrate_ and _quality_, but none of these seem to affect the image brightness.

Perhaps the problem is in encoders implemented in the GPU, which can't be adjusted through `picamera`?

On the figure below, the MJPEG and YUV images are far better quality than RGB and h264. Many of the objects clearly visible in MJPEG and YUV images are not visible at all in h264 and RGB images.

[image]

Below is the code used to capture the frames and to display them.
```python
from picamera import PiCamera
from picamera.array import PiRGBArray, PiYUVArray
from time import sleep
import numpy as np

camera = PiCamera(sensor_mode=4, framerate=30, resolution=(800, 608))

# Set gains and exposure to fixed values
camera.awb_mode = 'auto'
camera.exposure_mode = 'sports'
camera.iso = 800
camera.start_preview()
sleep(2)
gains = camera.awb_gains
camera.exposure_mode = 'off'
camera.awb_mode = 'off'
camera.awb_gains = gains
camera.shutter_speed = camera.exposure_speed
camera.image_denoise = False
camera.video_denoise = False
camera.stop_preview()

# Record mjpeg clip
camera.start_recording('test.avi', format='mjpeg')
sleep(5)
camera.stop_recording()

# Record h264 clip
camera.start_recording('test.h264', format='h264', 
                       quality=25)
sleep(5)
camera.stop_recording()

# Capture YUV frame from video port
with PiRGBArray(camera) as output:
    camera.capture(output, format='rgb', use_video_port=True)
    frame_rgb = output.array

# Capture YUV frame from video port
with PiYUVArray(camera) as output:
    camera.capture(output, format='yuv', use_video_port=True)
    frame_yuv = output.array

# Close camera
camera.close()

# Save RGB image as numpy file
np.save('test_rgb.npy', frame_rgb)
np.save('test_yuv.npy', frame_yuv)
```

And to display the captured frames
```python
import numpy as np
import cv2
import matplotlib.pyplot as plt

path_to_test_avi = 'test.avi'
path_to_test_h264 = 'test.h264'
path_to_test_rgb = 'test_rgb.npy'
path_to_test_yuv = 'test_yuv.npy'

use_frame_nr = 30

# Load MJPEG frame
reader = cv2.VideoCapture(path_to_test_avi)
for nr in range(use_frame_nr):
    ret, frame_mjpeg = reader.read()
reader.release()
# Convert frame from BGR to RGB
frame_mjpeg = cv2.cvtColor(frame_mjpeg, cv2.COLOR_BGR2RGB)

# Load h264 frame
reader = cv2.VideoCapture(path_to_test_h264)
for nr in range(use_frame_nr):
    ret, frame_h264 = reader.read()
reader.release()
# Convert frame from BGR to RGB
frame_h264 = cv2.cvtColor(frame_h264, cv2.COLOR_BGR2RGB)

# Load RGB frame
frame_rgb = np.load(path_to_test_rgb)

# Load YUV frame and convert to RGB for display
frame_yuv = np.load(path_to_test_yuv)
frame_yuv = cv2.cvtColor(frame_yuv, cv2.COLOR_YUV2RGB)

# Display frames
f, axes = plt.subplots(2, 2)
axes[0,0].imshow(frame_mjpeg)
axes[0,0].set_title('MJPEG')
axes[0,0].axis('off')
axes[0,1].imshow(frame_h264)
axes[0,1].set_title('h264')
axes[0,1].axis('off')
axes[1,0].imshow(frame_rgb)
axes[1,0].set_title('RGB')
axes[1,0].axis('off')
axes[1,1].imshow(frame_yuv)
axes[1,1].set_title('YUV')
axes[1,1].axis('off')
plt.show()
```

6by9, 2018-10-04:
Almost certainly due to mishandling of colour spaces / primaries.

There are several commonly used colour spaces - JFIF (JPEG), BT601 (SD video), and BT709 (HD video). These define the relevant matrices to use if you wish to convert from YUV to RGB. 

The ISP always produces YUV, and should be generating BT601 where luma samples range from 16-235. JFIF uses luma samples range from 0-255.
- The H264 encoder will correctly signal BT601 this in the bitstream.
- The conversion to RGB will also be abiding by BT601 matrices.
- MJPEG has no such metadata and will generally be interpreted as JFIF as it's based on JPEG.
- How your YUV decode converts to RGB is up to OpenCV, quite likely JFIF.

At the MMAL level you can tell the camera to signal JFIF instead via the ```color_space``` in the port format.
```
/** Unknown color space */
#define MMAL_COLOR_SPACE_UNKNOWN       0
/** ITU-R BT.601-5 [SDTV] */
#define MMAL_COLOR_SPACE_ITUR_BT601    MMAL_FOURCC('Y','6','0','1')
/** ITU-R BT.709-3 [HDTV] */
#define MMAL_COLOR_SPACE_ITUR_BT709    MMAL_FOURCC('Y','7','0','9')
/** JPEG JFIF */
#define MMAL_COLOR_SPACE_JPEG_JFIF     MMAL_FOURCC('Y','J','F','I')
```
I suspect it'll be set to AUTO, and you'll be looking to set it to ```MMAL_COLOR_SPACE_JPEG_JFIF```.

waveform80, 2018-10-04:
> I suspect it'll be set to AUTO, and you'll be looking to set it to `MMAL_COLOR_SPACE_JPEG_JFIF`.

If that's the default then yes, it'll be AUTO - I'm fairly sure picamera doesn't touch the `color_space` setting anywhere (largely due to me not knowing what to do with it before :). I'm tempted to mark this a bug in picamera and see if I can tweak the `color_space` to JFIF whenever an MJPEG encoder is attached to something (or allow the color_space to be specified as a parameter to start_recording).

However ... is it enough to tweak the `color_space` on the encoder's input port, or is this something that would need setting on the original output port of the camera component? If the latter my ill-fated decision to always stuff that splitter in the pipeline is going to complicate matters (at least for users that don't wire up their own pipeline down in `mmalobj`).

sanntann, 2018-10-10:
Thank you both for taking the time to respond.

I have not worked with the MMAL level before, but I tried to understand what is happening and to implement some (ugly) hard-coded fix, so I can move to using the JFIF color space for my project.

What I understand from your replies is that the `picamera` module is not currently set to specify `color_space`, which is automatically set in `libmmal` C-library based on the encoder type attached to the `picamera.mmalobj.MMALVideoPort` when used by the `picamera.mmalobj.MMALCamera`.

Indeed, I tried setting all the color codes in `mmal.py` to JFIF, but this had no effect. Changes were on lines 2256 - 2256 in `mmal.py`.
```python
MMAL_COLOR_SPACE_ITUR_BT601   = MMAL_FOURCC('YJFI')
MMAL_COLOR_SPACE_ITUR_BT709   = MMAL_FOURCC('YJFI')
MMAL_COLOR_SPACE_JPEG_JFIF    = MMAL_FOURCC('YJFI')
```

So I gave a shot at specifying the `color_code` for `picamera.mmalobj.MMALVideoPort` just before recording is started. I assume the correct place to do this would be earlier in the process of initialization, but I implemented it in the late stage because I understand the `color_code` may be overriden automatically at some stage and I don't properly understand the MMAL level to find out where the correct place is to set the `color_code`.

I added the following lines after line 397 in `encoders.py`:
```python
# Changes to hard-code JFIF color code
self.input_port._port[0].format[0].es[0].video.color_space = mmal.MMAL_FOURCC('YJFI')
self.input_port.commit()
self.output_port.copy_from(self.input_port)
self.output_port.commit()
# End of changes
```

So the `picamera.PiEncoder.start` method looks like this:
```python
    def start(self, output):
        """
        Starts the encoder object writing to the specified output.

        This method is called by the camera to start the encoder capturing
        data from the camera to the specified output. The *output* parameter
        is either a filename, or a file-like object (for image and video
        encoders), or an iterable of filenames or file-like objects (for
        multi-image encoders).
        """
        self.event.clear()
        self.exception = None
        self._open_output(output)
        with self.parent._encoders_lock:
            self.output_port.enable(self._callback)
            if self.DEBUG > 0:
                mo.print_pipeline(self.output_port)
            # Changes to hard-code JFIF color code
            self.input_port._port[0].format[0].es[0].video.color_space = mmal.MMAL_FOURCC('YJFI')
            self.input_port.commit()
            self.output_port.copy_from(self.input_port)
            self.output_port.commit()
            # End of changes
            self.parent._start_capture(self.camera_port)
```

This yields the following error when I attempt to record video in `'h264'` format:
```python
mmal: mmal_vc_port_info_set: failed to set port info (3:1): EINVAL
mmal: mmal_vc_port_set_format: mmal_vc_port_info_set failed 0x146ce20 (EINVAL)
```

Perhaps you could suggest where the correct place may be to specify color_space. I would really appreciate even an ugly hard-coded solution, just so I can continue with the project. I only use formats `'h264'`, `'mjpeg'` and `'yuv'`, so I'm guessing it wouldn't hurt if I set the `color_space` to always be JFIF.

6by9, 2018-10-10:
> If that's the default then yes, it'll be AUTO - I'm fairly sure picamera doesn't touch the color_space setting anywhere (largely due to me not knowing what to do with it before :). I'm tempted to [...]

> However ... is it enough to tweak the color_space on the encoder's input port, or is this something that would need setting on the original output port of the camera component? If the latter my [...]

I don't know how you propagate formats between components in the pipeline.
I'd go for the simple solution of **always** setting colour_space to JFIF on all components - at least that way the setting is consistent.

It's been a niggle that was raised back in the Broadcom days that the ISP isn't reprogrammed with the relevant conversion matrices to produce the desired colourspace, so this is purely a signalling thing, and things doing conversions to RGB are using inappropriate settings.

6by9, 2018-10-10:
You may be able to take the sledgehammer approach and set colourspace in ```commit``` https://github.com/waveform80/picamera/blob/master/picamera/mmalobj.py#L1112

I should add to the above that for MMAL_ENCODING_OPAQUE the colourspace is conveyed within the buffer, therefore it has to be set appropriately on the source component.

Please note that for H264 you are reliant on whatever is decoding the stream and converting it to RGB to look at the relevant bits in the headers.
For RGB I believe it is using video_splitter to do the conversion, and that should use the colourspace information appropriately.

commit 93a0808b (2018-11-17): Fix #512

Merge branch 'colorspace'
closed 2018-11-17 by waveform80