waveform80/picamera #516: Feature/Question: Would it be possible to capture at different specific framerates?

Esser420, 2018-10-11:
I'm working on a project that tries to make use of the PiCamera in many ways through a wrapper library. You can find the wrapper part [here](https://github.com/Esser420/PiCamStreamer).

Basically I found an use case where I would like to be able to record at different framerates at different ports.

So for example I would like to record to a file at a framerate of 32 but only stream to the network at maybe 16fps.

So for this I'd imagine using the "framerate" keyword in the "start_recording" method.
E.g.:
```python
start_recording("h264", port = 0, framerate = 30)
start_recording("h264", port = 1, framerate = 16)
```

6by9, 2018-10-13:
Your source code says 30fps and 16fps which certainly is not possible.
If you mean 32 and 16 as in your text then it's plausible, however there's no easy way to drop every other frame within a pipeline so I doubt Picamera will let you.

If you switched to C and used MMAL directly yourself then you could set your own buffer callbacks that only sent every other frame to the encoder, but it would be a custom app.

waveform80, 2018-11-09:
Yup, no way to manage this within the high-level picamera API. Theoretically possible via the mmalobj layer, but that's still largely undocumented (going to try and improve that for 1.14, but time constraints as ever...). Anyway, this definitely won't/can't be added in the way envisaged here, sorry.

closed 2018-11-09 by waveform80