waveform80/picamera #522: Definde camera.shutter_speed does not match with queried camera.exposure_speed

snowman907, 2018-10-26:
Defined camera.shutter does not match with queried camera.exposure_speed
Below is my code where shutter_speed is defined as 10 seconds.
But when querying the camera object with camera.exposure_speed the result is 0. 
That does not make sense!

`with PiCamera(resolution=(1024,768),framerate=Fraction(1,6),sensor_mode=3) as camera:
            
            exp_sec=int('10')
            camera.shutter_speed=exp_sec*10**6 # micros
            sleep(30)
            print('camera.shutter_speed='+str(camera.shutter_speed))
            print('camera.exposure_speed:'+str(camera.exposure_speed))                
            camera.iso=1600 #100-1600
            camera.exposure_mode='off' # lock all setting parameters
            fn_png=str(time.strftime("%Y-%m-%d-%H-%M-%S"))+'.png
            camera.capture(fn_png, format='png')'
In response:
```
camera.shutter_speed=9999959
camera.exposure_speed=0
```

snowman907, 2018-11-06:
Anyone can help me with this one?

snowman907, 2018-11-11:
Guys, I am still looking for someone to shine a light on this, thanks!

mikevanis, 2022-01-26:
Hello! Any desire / clues for this to be fixed? Happy to help in whatever way I can, as we use this feature quite a lot in [My Naturewatch Camera](https://mynaturewatch.net/) :)