waveform80/picamera #523: Invalid H.264 profile extended

miguelangelalba, 2018-10-28:
I'm doing a raspberry streaming project for the university.
In the documentation I saw that I could use the "extended" encoding profile but it is coming out as an invalid argument. Can it be a bug?

`    camera.start_recording(filename, format='h264', quality = 10, level = 4,profile = 'extended')
`

6by9, 2018-10-30:
Extended profile encoding is not supported by the firmware.
There is an enum specified in OpenMax and MMAL for them as they are generic APIs, but setting extended profile will fail on the video_encode component.

There is ```OMX_IndexParamVideoProfileLevelQuerySupported``` / ```MMAL_PARAMETER_SUPPORTED_PROFILES``` to query the profiles supported by a component.

NB ```quality``` will mess up bitrate control on h264 encoding. It fixes the min and max quantisation values in the encoder, so is almost a fixed quality method rather than any form of rate control. More normally you'd set the bitrate and allow the codec to control quantisation levels.

miguelangelalba, 2018-10-30:
Thanks for the support. It took me to confusion the documentation :) 
[image]

waveform80, 2018-11-09:
Looks like it's at least a documentation issue; I'll get that fixed for 1.14

commit 016b7185 (2018-11-15): Fix #523

Extended profile for H.264 is not supported by the firmware
closed 2018-11-15 by waveform80