waveform80/picamera #525: PiCamera under docker

rimokhy, 2018-10-29:
Hey,

I've been using PiCamera through docker on my raspberry pi 3. The container run pretty fine pulling base image from `resin/raspberry-pi-python:3`
However, I cannot allow my image to take 500Mo for the resin image, and another 500Mo for the final build. That's why I decided to go for an alpine build and install everything that is needed on my own.

First error, at picamera installing, it could not find `libbcm_host.so`, and later, `libmmal.so`.
I have solved those behaviors by adding the alpine package `raspberrypi-libs` and adding a symbolic link 
`ln -sf /opt/vc/lib/libbcm_host.so /usr/lib/libbcm_host.so`
`ln -sf /opt/vc/lib/libmmal.so /usr/lib/libmmal.so`
This fix passes the build phase. 

However, I have been stuck, at runtime this time by the following error :
`File "./wsgi.py", line 1, in <module>
from app import app
File "./app.py", line 15, in <module>
camera = PiCamera()
File "/usr/local/lib/python3.6/site-packages/picamera/camera.py", line 367, in __init__
with mo.MMALCameraInfo() as camera_info:
File "/usr/local/lib/python3.6/site-packages/picamera/mmalobj.py", line 2346, in __init__
super(MMALCameraInfo, self).__init__()
File "/usr/local/lib/python3.6/site-packages/picamera/mmalobj.py", line 633, in __init__
prefix="Failed to create MMAL component %s" % self.component_type)
File "/usr/local/lib/python3.6/site-packages/picamera/exc.py", line 184, in mmal_check
raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.camera_info': Function not implemented
`
I feel like I can see the light at the end of the tunnel. Any help would be highly appreciated.

Also, I have been describing the steps I have taken, that will hopefully help newcomers that encounter the same issues.

RobertLucian, 2018-11-02:
Right, I hit this wall not long ago. I spent hours figuring out what packages are required for it to work. Turns out there are quite a lot:
https://github.com/RobertLucian/LED-QA-GiggleBot/blob/master/Dockerfile#L26-L34

Out of all these packages, you won't need `python3 python3-dev python3-pip wiringpi`. 

Also, when running the container make sure to give access to the camera device. Personally, I just use `--privileged` when I'm not worried about security.

rimokhy, 2018-11-03:
When you do worry about security, i advice you allow docker only on the prefered device. For picamera, you need to point it out to /dev/vchiq with the device option.

Thanks for the tips!