waveform80/picamera #527: Setting the system date/time kills the camera!

acu192, 2018-11-06:
**My stetup:**
- Raspberry Pi 3 Model B  (not B+)
- Raspbian Stretch Lite (build 2018-10-09)
- Python 3.6
- picamera 1.13

**Steps to reproduce:**

Step 1: Run this following script (in its own terminal, so you can see it when it dies)

```python
from picamera import PiCamera
from picamera.array import PiRGBArray
import time

width, height = 320, 240

camera = PiCamera(resolution=(width, height), framerate=8)
array = PiRGBArray(camera, size=(width, height))

time.sleep(1.0)

for _ in camera.capture_continuous(array, format='rgb', use_video_port=True):
    frame = array.array
    array.truncate(0)
    print('Capture frame with shape', frame.shape)

camera.close()
```

Step 2: Run the following commands (in a different terminal, of course).

```bash
> sudo systemctl stop systemd-timesyncd.service    # otherwise the next command won't work
> sudo date -s "Tue Nov 6 02:41:35 UTC 2017"        # set to a time in the past
> sudo date -s "Tue Nov 6 02:41:35 UTC 2018"       # set the time back to "now"
```

You should see the python script from step 1 die now.

I suppose the timeout mechanism that picamera uses gets thrown off when the system time is changed.

Unfortunately I hit this error all the time (I only recently figured out it had to do with the time being set). Reason: On-boot I have a process spin up and start using the camera. Then my Pi gets Wifi access a few seconds later, and then the background service which queries the NTP server wakes up and gets the current time and sets it, and then the camera dies.

Btw, this library is great. If you give me a little direction, I'd be more than happy to help fix this bug. Thank you!

acu192, 2018-11-06:
Maybe the issue here (described above) is the underlying cause of this other issue: https://github.com/waveform80/picamera/issues/242  ??

RDReavis, 2021-01-01:
Confirmed that this is still an issue and impacts my project.

The firmware issue linked to #242 that might have resolved this, raspberrypi/firmware#609, appears to have been closed without a fix. Is the proper action to open a new firmware issue?