waveform80/picamera #529: Long exposures of >8s hang

jaimemarijke, 2018-11-14:
picamera V1.13 with camera module V2.1 on Pi 3 B+.

I've been trying to test out taking long exposures, and seem to have run into a bug where the capture hangs if the exposure is greater than ~8s.

My code:
```
from fractions import Fraction
from time import sleep, time

import picamera
import picamera.array

with picamera.PiCamera(framerate=Fraction(1,9), sensor_mode=2) as camera:
    camera.iso = 100
    camera.shutter_speed = 9 * 1000 * 1000
    camera.exposure_mode = 'off' 
    camera.awb_mode = 'off'

    with picamera.array.PiBayerArray(camera) as stream:
        sleep(2)
        start_time = time()
        camera.capture(stream, 'jpeg', bayer=True)
        print('Elapsed time:', time() - start_time)

        # Setting the framerate back above 1 allows camera.close() to actually succeed
        # https://github.com/waveform80/picamera/issues/528
        camera.framerate = 1
```

Elapsed time measurements:
```
# framerate=Fraction(1,2), shutter_speed = 2 * 1000 * 1000 --> ~13s
# framerate=Fraction(1,6), shutter_speed = 6 * 1000 * 1000 --> ~42s
# framerate=Fraction(1,7), shutter_speed = 7 * 1000 * 1000 --> ~49s
# framerate=Fraction(1,8), shutter_speed = 8 * 1000 * 1000 --> ~55s
# framerate=Fraction(1,9), shutter_speed = 9 * 1000 * 1000 --> hangs? (gave up after ~5-10 min)
# framerate=Fraction(1,10), shutter_speed = 10 * 1000 * 1000 --> hangs?
```

These tests lead me to conclude that it takes about 7x the shutter_speed to actually capture a long exposure (Incidentally, I'd love to understand why).

I noticed that this seems to start hanging suspiciously around 60s elapsed time. The top bullet in your [v1.13 changelog notes](https://picamera.readthedocs.io/en/latest/changelog.html?#release-1-13-2017-02-25) makes me suspect that at shutter_speeds greater than 8s, we start to hit that timeout, but that PiCamera doesn't handle the timeout well and instead hangs.

techhead2, 2019-01-27:
My test show that when the framerate becomes slower than 1fps the camera starts hanging. For low-light conditions the framerate should be:
`camera.framerate = Fraction(1, 6)`
but as soon as I get it under
`camera.framerate = Fraction(1, 1)`
the camera hangs and only a reboot helps.

I would badly need this option. Please try to find this problem. My env:
Raspberry with
`> uname -a`
`Linux xyz 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux`
`> pip show picamera`
`Name: picamera`
`Version: 1.13`
`Summary: A pure Python interface for the Raspberry Pi camera module.`
`...`

techhead2, 2019-02-04:
Yes, I've seen it and it works. Nice job Jamie, although a fix would be even nicer :-). Your question about the expo time is also valid and annoying. I'm doing long expos and its a pain to await for them. Would be nice to have a fix for it, too.

tatref, 2019-09-26:
I'm also trying to get long exposures to work. I also noted the increasing timings.

Since the C tools (raspistill) are also impacted, I believe the issue lies in the firmware: https://github.com/raspberrypi/firmware/issues/1233

Also, as explained in https://github.com/waveform80/picamera/issues/528, I think you want `sensor_mode=3` (see the camera modes https://picamera.readthedocs.io/en/release-1.10/fov.html#camera-modes)

Thanks for the `framerate=1` workaround!