waveform80/picamera #539: How to install picamera on mac so that linters don't complain about missing imports

Esser420, 2019-01-07:
Basically I would like pylint not to complain about the missing picamera package but still suggest auto-completion.

I do understand that the library itself won't work on mac, but is there a way to somehow only download the source?

Flightkick, 2019-03-26:
@LizardM4 commented the same on another issue. https://github.com/waveform80/picamera/issues/404#issuecomment-332461857

I found your issues while looking for a solution to this myself.

Warning: picamera only works on raspberry pi systems. This solution allows you to install picamera on a different OS so you can use the library for code completion in your IDE. The library will however be nonfunctional.

Optional: If you are using Python venv (Virtual Environment) first activate your environment by opening a terminal and executing the `activate` or `activate.bat`

On Windows, execute: set READTHEDOCS=True
On Unix based systems, execute: export READTHEDOCS=True

Now execute: pip install picamera

The library should install without issues.

Optional: If you are using Python venv, you can exit your virtual environment by running `deactivate`

arpit7714, 2019-10-10:
worked

kinite-gp, 2023-01-16:
The problem is fixed and it is installed
But I'm curious to know what it is and what it does.
`set READTHEDOCS=True`
Can you give an explanation?

Flightkick, 2023-03-02:
@kinite-gp Good question, had to look it up since it's been a while.
Setting the variable to `True` skips the check which validates if the installation is running on a Raspbery Pi.

This is the relevant code for the latest release:
https://github.com/waveform80/picamera/blob/e18d70989eb6c3bf19f4cea859c7401bea3b9048/setup.py#L97-L114

It seems that the current `master` branch doesn't contain this check anymore but the latest release (`1.13`) does.