waveform80/picamera #540: Install on 64bit OS (not raspian)

Truemedia, 2019-01-19:
Hi,
I'm using a 64 bit os for my raspberry pi so I can use libraries for projects where the 32bit versions would normally fallover (eg mongodb 2gb limit).

I want to use the camera but Iv'e had massive difficulty getting it on any 64 bit os (worked out of the box fine for 32bit raspian).

I'm using this currently as my os: https://github.com/bamarni/pi64

I tried installing this even though i knew it would probably fail, by installing via pip. During pip install it errors out with this message: ValueError: Unable to determine if this system is a Raspberry Pi

Just wondering any futher info you might have on any alternative routes I could get towards this working. I'm wanting to work on some advanced computer vision projects but a 32bit os just won't cut it.

6by9, 2019-01-21:
PiCamera is just a wrapper over the top of the MMAL API for accessing the GPU functionality.

MMAL does not support 64bit OSes at present, therefore PiCamera can't.

You should be able to use the V4L2 driver under a 64bit OS. I haven't tried it personally, but I think all the hurdles stopping it have been resolved. As an API V4L2 is more restricted in functionality than MMAL, so you won't be able to do everything using it - sorry, that's the way it is.

Truemedia, 2019-01-24:
I will have a look into that thanks

domyam97, 2020-05-22:
Since userland got updated for the 64 bit arch support in https://github.com/raspberrypi/userland/pull/586 I've been able to get picamera running on Ubuntu 20.04 Server on a RPi 3B+. 

[This guide](https://larrylisky.com/2016/11/24/enabling-raspberry-pi-camera-v2-under-ubuntu-mate/) had it pretty much nailed down step by step except I had to modify the CMakelists in **userland/interface/mmal** before running `./buildme --aarch64`. 

I added a `SET( CMAKE_SHARED_LINKER_FLAGS(-Wl, --no-as-needed))` before the `target_link_libraries(...)`. If I didn't do this I would get an error as described in #334. 

You also need to add your user to the video group and make sure that `/dev/vchiq` is owned by the video group.

Elly-78, 2020-09-01:
> Since userland got updated for the 64 bit arch support in [raspberrypi/userland#586](https://github.com/raspberrypi/userland/pull/586) I've been able to get picamera running on Ubuntu 20.04 Server on a RPi 3B+.
> 
> [This guide](https://larrylisky.com/2016/11/24/enabling-raspberry-pi-camera-v2-under-ubuntu-mate/) had it pretty much nailed down step by step except I had to modify the CMakelists in **userland/interface/mmal** before running `./buildme --aarch64`.
> 
> I added a `SET( CMAKE_SHARED_LINKER_FLAGS(-Wl, --no-as-needed))` before the `target_link_libraries(...)`. If I didn't do this I would get an error as described in #334.
> 
> You also need to add your user to the video group and make sure that `/dev/vchiq` is owned by the video group.

Hi, really need some help. I am trying to follow your instructions but keep getting :
 `~/userland/build/raspberry/release ~/userland
./buildme: line 24: cmake: command not found
make: *** No targets specified and no makefile found.  Stop.
[sudo] password for pi: 
make: *** No rule to make target 'install'.  Stop.
~/userland
 `

what am i doing wrong?
thanks
P.S. I tried on both ubuntu mate 64 bit and the beta rpi OS 64

quajak, 2020-10-06:
You have to install cmake

3iljw, 2020-11-03:
> Since userland got updated for the 64 bit arch support in [raspberrypi/userland#586](https://github.com/raspberrypi/userland/pull/586) I've been able to get picamera running on Ubuntu 20.04 Server [...]

Hi, I have same system version with you but on rpi 4.
But I can't successfully boot anymore after I do 'rpi-update'.
it's final message on screen and then the screen turned blank
[image]

kpoman, 2021-07-16:
Same problem here. Apparently, mmal for 64 bits is buggy (race conditions and crashs). There are binaries one can download, but with this above problem (vc.camera_info). This [link](https://zengliyang.wordpress.com/2021/01/04/raspberry-pi-4b-ubuntu-20-04-camera/) says we should add SET(CMAKE_SHARED_LINKER_FLAGS "-Wl,--no-as-needed") to the userland package that builds mmal, but it doesnt work (i.e. doesnt build) on 64bit. People suggest using libcamera but it makes picamera unusable as it takes full control of the device. So I gues we are stuck on 64 bits for now. Unfortunately !

6by9, 2021-07-16:
https://github.com/raspberrypi/userland/issues/688
MMAL is not supported on 64 bit OSes.

Ubuntu have reinstated the 64bit MMAL patches in their distro. It will not be reinstated in the main Raspberry Pi userland repo until the cause of the issues have been identified and fixed as it causes issues with 32bit other MMAL use cases (video decoder in particular).

kpoman, 2021-07-16:
> [raspberrypi/userland#688](https://github.com/raspberrypi/userland/issues/688)
> MMAL is not supported on 64 bit OSes.
> 
> Ubuntu have reinstated the 64bit MMAL patches in their distro. It will not be reinstated in the main Raspberry Pi userland repo until the cause of the issues have been identified and fixed as it causes issues with 32bit other MMAL use cases (video decoder in particular).

I am using ubuntu server 64 bit, installed both libraspberrypi0 and libraspberrypi-bin. These packages provide the following files:

```
ubuntu@rpi3:~/Dev/userland$ dpkg -L libraspberrypi-bin
/.
/usr
/usr/bin
/usr/bin/dtmerge
/usr/bin/dtoverlay
/usr/bin/dtoverlay-post
/usr/bin/dtoverlay-pre
/usr/bin/raspistill
/usr/bin/raspivid
/usr/bin/raspividyuv
/usr/bin/raspiyuv
/usr/bin/tvservice
/usr/bin/vcgencmd
/usr/bin/vcmailbox
/usr/lib
/usr/lib/userland
/usr/lib/userland/mmal_vc_diag
/usr/lib/userland/vchiq_test
/usr/share
/usr/share/doc
/usr/share/doc/libraspberrypi-bin
/usr/share/doc/libraspberrypi-bin/copyright
/usr/share/lintian
/usr/share/lintian/overrides
/usr/share/lintian/overrides/libraspberrypi-bin
/usr/share/man
/usr/share/man/man1
/usr/share/man/man1/dtmerge.1.gz
/usr/share/man/man1/dtoverlay.1.gz
/usr/share/man/man1/dtparam.1.gz
/usr/share/man/man1/raspistill.1.gz
/usr/share/man/man1/raspivid.1.gz
/usr/share/man/man1/raspividyuv.1.gz
/usr/share/man/man1/raspiyuv.1.gz
/usr/share/man/man1/tvservice.1.gz
/usr/share/man/man1/vcgencmd.1.gz
/usr/share/man/man1/vcmailbox.1.gz
/usr/share/man/man7
/usr/share/man/man7/raspicam.7.gz
/usr/share/man/man7/raspiotp.7.gz
/usr/share/man/man7/raspirev.7.gz
/usr/share/man/man7/vcmailbox.7.gz
/usr/bin/dtparam
/usr/share/doc/libraspberrypi-bin/LICENCE
/usr/share/doc/libraspberrypi-bin/changelog.Debian.gz
ubuntu@rpi3:~/Dev/userland$ dpkg -L libraspberrypi0
/.
/lib
/lib/udev
/lib/udev/rules.d
/lib/udev/rules.d/10-local-rpi.rules
/usr
/usr/lib
/usr/lib/aarch64-linux-gnu
/usr/lib/aarch64-linux-gnu/libbcm_host.so
/usr/lib/aarch64-linux-gnu/libcontainers.so
/usr/lib/aarch64-linux-gnu/libdebug_sym.so
/usr/lib/aarch64-linux-gnu/libdtovl.so
/usr/lib/aarch64-linux-gnu/libmmal.so
/usr/lib/aarch64-linux-gnu/libmmal_components.so
/usr/lib/aarch64-linux-gnu/libmmal_core.so
/usr/lib/aarch64-linux-gnu/libmmal_util.so
/usr/lib/aarch64-linux-gnu/libmmal_vc_client.so
/usr/lib/aarch64-linux-gnu/libvchiq_arm.so
/usr/lib/aarch64-linux-gnu/libvcos.so
/usr/lib/aarch64-linux-gnu/libvcsm.so
/usr/lib/aarch64-linux-gnu/plugins
/usr/lib/aarch64-linux-gnu/plugins/reader_asf.so
/usr/lib/aarch64-linux-gnu/plugins/reader_avi.so
/usr/lib/aarch64-linux-gnu/plugins/reader_binary.so
/usr/lib/aarch64-linux-gnu/plugins/reader_flv.so
/usr/lib/aarch64-linux-gnu/plugins/reader_metadata_id3.so
/usr/lib/aarch64-linux-gnu/plugins/reader_mkv.so
/usr/lib/aarch64-linux-gnu/plugins/reader_mp4.so
/usr/lib/aarch64-linux-gnu/plugins/reader_mpga.so
/usr/lib/aarch64-linux-gnu/plugins/reader_ps.so
/usr/lib/aarch64-linux-gnu/plugins/reader_qsynth.so
/usr/lib/aarch64-linux-gnu/plugins/reader_raw_video.so
/usr/lib/aarch64-linux-gnu/plugins/reader_rcv.so
/usr/lib/aarch64-linux-gnu/plugins/reader_rtp.so
/usr/lib/aarch64-linux-gnu/plugins/reader_rtsp.so
/usr/lib/aarch64-linux-gnu/plugins/reader_rv9.so
/usr/lib/aarch64-linux-gnu/plugins/reader_simple.so
/usr/lib/aarch64-linux-gnu/plugins/reader_wav.so
/usr/lib/aarch64-linux-gnu/plugins/writer_asf.so
/usr/lib/aarch64-linux-gnu/plugins/writer_avi.so
/usr/lib/aarch64-linux-gnu/plugins/writer_binary.so
/usr/lib/aarch64-linux-gnu/plugins/writer_dummy.so
/usr/lib/aarch64-linux-gnu/plugins/writer_mp4.so
/usr/lib/aarch64-linux-gnu/plugins/writer_raw_video.so
/usr/lib/aarch64-linux-gnu/plugins/writer_simple.so
/usr/share
/usr/share/doc
/usr/share/doc/libraspberrypi0
/usr/share/doc/libraspberrypi0/LICENCE
/usr/share/doc/libraspberrypi0/changelog.Debian.gz
/usr/share/doc/libraspberrypi0/copyright
/usr/share/lintian
/usr/share/lintian/overrides
/usr/share/lintian/overrides/libraspberrypi0
ubuntu@rpi3:~/Dev/userland$ 

```
raspistill works:
`ubuntu@rpi3:~/Dev/userland$ raspistill -o test.jpg`
however, picamera doesnt:
```
ubuntu@rpi3:~/Dev/userland$ python3 -c "import picamera; picamera.PiCamera();"
mmal: mmal_component_create_core: could not find component 'vc.camera_info'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/picamera/camera.py", line 367, in __init__
    with mo.MMALCameraInfo() as camera_info:
  File "/usr/local/lib/python3.9/dist-packages/picamera/mmalobj.py", line 2346, in __init__
    super(MMALCameraInfo, self).__init__()
  File "/usr/local/lib/python3.9/dist-packages/picamera/mmalobj.py", line 631, in __init__
    mmal_check(
  File "/usr/local/lib/python3.9/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.camera_info': Function not implemented
ubuntu@rpi3:~/Dev/userland$
```
So, apparently ubuntu did compile the 64 bit version, but forgot to add that suggested cmake directive ( CMAKE_SHARED_LINKER_FLAGS(-Wl, --no-as-needed)). 
Unofrtunately, getting the repo and compiling doesnt compile mmal stuff if configured with ./buildme --arch64 ... it is apparently disabled here:
```
if(ARM64)
   set(BUILD_MMAL FALSE)
   set(BUILD_MMAL_APPS FALSE)
else()
   set(BUILD_MMAL TRUE)
   set(BUILD_MMAL_APPS TRUE)
endif()
```
Do you suggest going to ubuntu 64 bits for raspberry and asking if they could just add the cmake directive, or try to compile locally ?

kpoman, 2021-07-16:
Ok, I think I got it working. Did this:
```
git clone https://github.com/raspberrypi/userland
cd userland
git revert f97b1af1b3e653f9da2c1a3643479bfd469e3b74
git revert e31da99739927e87707b2e1bc978e75653706b9c
```
then edited userland/interface/mmal/CMakeLists.txt, and added the directive:
```
# We support building both static and shared libraries
if (NOT DEFINED LIBRARY_TYPE)
set(LIBRARY_TYPE SHARED)
endif (NOT DEFINED LIBRARY_TYPE)

add_definitions(-Wall -Werror)

add_library(mmal SHARED util/mmal_util.c)

add_subdirectory(core)
add_subdirectory(util)
add_subdirectory(vc)
add_subdirectory(components)
if(NOT ARM64)
   add_subdirectory(openmaxil)
   add_subdirectory(client)
endif()

SET(CMAKE_SHARED_LINKER_FLAGS "-Wl,--no-as-needed")
target_link_libraries(mmal mmal_core mmal_util mmal_vc_client vcos mmal_components)
```
....
then ran ./buildme --aarch64
It built everything to /opt/vc/ ... I then copied all built libs: sudo cp /opt/vc/lib/*.so /usr/lib/aarch64-linux-gnu/
Finally, I tested (I had some debug prints I have put there):
```
ubuntu@rpi3:~/Dev/userland2/userland$ python3 -c "import picamera; print(picamera); picamera.PiCamera();"
<CDLL 'libmmal.so', handle 2451dbf0 at 0xffffac0f83d0>
<module 'picamera' from '/usr/local/lib/python3.9/dist-packages/picamera/__init__.py'>
mmal_component_create
b'vc.camera_info'
<picamera.mmal.LP_MMAL_COMPONENT_T object at 0xffffabd05640>
postcheck
mmal_component_create
b'vc.ril.camera'
<picamera.mmal.LP_MMAL_COMPONENT_T object at 0xffffabd332c0>
postcheck
mmal_component_create
b'vc.null_sink'
<picamera.mmal.LP_MMAL_COMPONENT_T object at 0xffffabd33140>
postcheck
mmal_component_create
b'vc.ril.video_splitter'
<picamera.mmal.LP_MMAL_COMPONENT_T object at 0xffffabd330c0>
postcheck
ubuntu@rpi3:~/Dev/userland2/userland$
```
Will test more and get posted.

./buildme --aarch64

6by9, 2021-07-16:
Feel free to revert those commits and build for yourself, but at the present time you will not be getting support for that configuration from Pi Towers.
When resources allow we will investigate, but it is not a high priority.

surak, 2022-02-16:
It would make sense for this project to move to v4l2, then? It's the one used by a lot of other projects.