waveform80/picamera #541: framerate_range + sps_timing option results in output file of 500fps

vincentneo, 2019-01-22:
Code that I tested with:
```with picamera.PiCamera() as camera:
    camera.resolution = (1920, 1080)
    camera.framerate_range = (10, 30)
    time.sleep(5)                               
    camera.start_recording("testvid.h264", bitrate = 8000000, quality = 30, format='h264', sps_timing = True)
    camera.wait_recording(10)
    camera.stop_recording()
```
It appears that when `sps_timing = True` , the output file would have 500fps, as when played, it will be too fast, and when placed into MP4Box, it would also show this:
 `AVC-H264 import - frame size 1920 x 1080 at 500.000 FPS`
Which looks wrong as well. 
Having `sps_timing = False` will result in a 25fps file which I kinda expected. 
The version I am testing with is the current master branch clone. It might be related to issue #502 ?

6by9, 2019-01-24:
Your options conflict (although I wouldn't immediately expect to get 500fps).

You're asking for a dynamically changing frame rate in the range 10-30fps, and yet want a single number in the SPS VUI for timing. The only valid number to insert in there is 0, or omit the section.

I suspect the 500fps is MP4Box not looking at the full header.
Reading the header spec, when the framerate is variable
- timing_info_present will be 1
- num_units_in_tick will be 1
- time_scale will be 1000 (hard coded number from somewhere)
- fixed_frame_rate_flag will be 0

If the frame rate is fixed, then
- time_scale will be 2 * framerate
- fixed_frame_rate_flag will be 1

Treating the 1000 in the same way as fixed frame rate would give you 500fps, but I'm expecting it to be effectively invalid without fixed_frame_rate_flag set.
I'm not enough of an expert on H264 bitstreams to say what the correct interpretation/settings should be under these conditions, but as I said above it's going to be a bogus number anyway. What number are you expecting?

vincentneo, 2019-01-24:
I needed to use a variable frame rate in this case, and I thought that since sps_timing is supposed to to have the frame rate in the header, it would be better. I was surprised to see 500fps, and playing back the .h264 video in VLC was very fast as well.

I expected to see a number in between the range of 10-30fps, as that was what was coded.

6by9, 2019-01-24:
If you want to use variable frame rate, then you need to use the frame timestamps. Anything else is wrong.

The codec doesn't know the frame rate range is defined as 10-30fps - it gets told 0 for variable.

The only logical thing I can do is amend the firmware to not insert the timing info when frame rate is 0, but I will get the view of a colleague who has more knowledge of H264 than me.

6by9, 2019-01-24:
I've asked my colleague and he doesn't have any insight.

In the abscence of any better ideas, the only change I'd consider is to not insert the timing info at all if the frame rate has not been provided (```timing_info_present``` will be 0).
You still get a few of the extra fields from the VUI, but it will be up to the decoder to guess at a frame rate.
Overall I fail to see why you're asking for sps_timing in the first place when using a dynamic frame rate.

vincentneo, 2019-01-27:
Ok. I guess I'll disable that, and use it as is, as this wasn't a problem without it.

@6by9 Thanks for the explanations!

closed 2019-01-27 by vincentneo