waveform80/picamera #542: Unable to view JPEGs after still image capture

Bobberty, 2019-01-29:
Debian Buster Linux 4.20.4 running on Rpi3B with Python 3.7 and PICamera 1.13 from both PIP and Git/Master (Same result)

While running the following code:

`from time import sleep
from picamera import PiCamera

camera = PiCamera()
camera.resolution = (1024, 768)
camera.start_preview()
sleep(2)
camera.capture('foo.jpg')`
`

The file is saved as "foo.jpg".
But the first 32 bytes of the file are cleared.
I'm unable to view the image from any application.

6by9, 2019-01-30:
> Debian Buster Linux 4.20.4

Wow, that's brave. Where have you got your 4.20 kernel from, and with what defconfig and device tree?

RPT are mainly looking at 4.19 as the latest LTS release, so 4.20 may well have issues.
There were also changes to the device tree bindings for getting the cache size information correct in VCHIQ (IIRC the compatible string changed from bcm2835 to bcm2836). If that is incorrect then you will get funny issues with partial corruption.

Can you reproduce with a kernel from https://github.com/raspberrypi/linux? Using '''sudo BRANCH=next rpi-update''' will get you the latest 4.19 kernel build.

Bobberty, 2019-01-30:
Further testing shows that raspistill works fine for JPEG images

I am using https://github.com/raspberrypi/linux  Branch=rpi-4.20.y.
More specifically, I'm using https://github.com/drtyhlpr/rpi23-gen-image to create the images.

I will rebuild using 4.19 and report my findings.

6by9, 2019-01-30:
https://github.com/raspberrypi/linux/commit/c683db8860a80562a2bb5b451d77b3e471d24f36#diff-81c872ae515a5e795b7647726768048f is in 4.20 and 5.0, but the associated DT change appears not to be in either https://github.com/anholt/linux/commit/499770ede3f829e80539f46b59b5f460dc327aa6

6by9, 2019-01-30:
You say that raspistill worked, but it'll be chance. It depends on exactly the alignment and cached status of the bytes in the buffer used to carry the JPEG from the GPU to ARM.

PR created to get the DT changes on the rpi-4.20.y branch.

Bobberty, 2019-01-30:
rpi-4.19.y works.  Thanks for pushing the merge.

6by9, 2019-01-30:
OK, almost certainly that missing DT change then.

TBH I don't know what the plan is with 4.20 (or 5.0). We tend to support the LTS releases otherwise we are having to rework the downstream patches almost constantly. Any issues on the intermediate releases will be treated as a lower priority. Unless you have a specific need for 4.20 I'd advise sticking to 4.19.

6by9, 2019-02-04:
Please close the issue if your problem is resolved.

closed 2019-02-07 by Bobberty