waveform80/picamera #546: picamera capture hang issue

cfdcfc, 2019-02-27:
Hi,

I am trying to record video with two ports with different resize value, and I want to capture image while recording.
However the picamera will hang when capturing the image.

The issue can be reproduced with below codes.

```
from picamera import PiCamera

camera = PiCamera()

camera.resolution = (1920,1080)
camera.framerate = 25
print("start recording on port 1")
camera.start_recording(output='1.h264', splitter_port=1, resize=(960,540))
print("start recording on port 2")
camera.start_recording(output='2.h264', splitter_port=2, resize=(300,300))

camera.wait_recording(5)
print("start capturing")
camera.capture('test.jpg', use_video_port=True)
print("captured to test.jpg")
```

xxxSon1cxxx, 2019-03-22:
I had a similiar problem after firmware/kernel upgrade to the latest version using **rpi-update**. I tried to find on which firmware this problem starts and found that on firmware until 4.9.61 everything is fine. So you can try to downgrade your firmware to this version https://github.com/Hexxeh/rpi-firmware/commit/ff196a8a73de039aeebc81c7c1f1e4a6580db11c

6by9, 2019-03-25:
@xxxSon1cxxx Are you saying that https://github.com/Hexxeh/rpi-firmware/commit/23b1614bb794dfce586f1300da75198befa4951d fails for you then? None of those changes would touch camera, resize, or splitter.

Please be very aware that the 4.9 is now unsupported by Raspberry Pi (last touched over a year ago), therefore security issues are your own problem. 4.14 should be deprecated soon as we switch to 4.19.

xxxSon1cxxx, 2019-03-25:
> Are you saying that [Hexxeh/rpi-firmware@23b1614](https://github.com/Hexxeh/rpi-firmware/commit/23b1614bb794dfce586f1300da75198befa4951d) fails for you then? None of those changes would touch camera, resize, or splitter.

Yes, on 4.9.62 my python script, which also record video in two ports and with working camera preview, hangs on string 
`self.camera.capture( imageData, 'bmp', use_video_port=True, resize=(self.testWidth, self.testHeight) ) `

dneunkirch, 2020-10-02:
Hi @xxxSon1cxxx how did you solve this issue. I can reproduce this (https://github.com/waveform80/picamera/issues/649). If I use an older kernel (4.9.61) it works.

flomine, 2020-10-08:
Hi, same problem here with Pi 3B and camera V2 on Buster.

```
camera.capture('test.jpg', use_video_port=True)
File "/home/pi/.virtualenvs/test/lib/python3.7/site-packages/picamera/camera.py", line 1423, in capture 'Timed out waiting for capture to end')
picamera.exc.PiCameraRuntimeError: Timed out waiting for capture to end
```

matthiasharrer, 2022-05-03:
Hey, did anybody solve or workaround this issue in the meantime? 

I think I am experiencing a similar issue on Raspberry Pi 4 and v2 camera.
Strangely it seems to only affect the hardware revision 1.4.

I tested both buster and bullseye (with legacy camera stack) with the same issue.
If I however switch to `use_video_port=False` it does not hang (but the video I am recording at the same time pauses for a few frames).

Maybe it's some synchronization problem only occuring with the `use_video_port option`?