waveform80/picamera #552: Capturing preview with low res but still at full resolution

mkoppanen, 2019-03-23:
I’ve been testing a photo booth use case and roughly the requirements are:

Run preview at lower resolution to achieve reasonable FPS. When capture is requested (there is a countdown and then a still capture) I would like to capture at max res.

I am using capture_continuous and initialise the camera with 1280x720 resolution and 30 frame rate. I tried first initialising using full res and using the resize for preview but FPS drops to somewhere around 0.5 - 1.

I have done a couple tests where I capture a still while using video port for preview but setting resolution before still capture (without stopping capture_continuous) results into whole pi crashing and needing hard reboot.

Is there a way where I can use video port around 1280x720 and capture stills at max resolution? 

I can post code samples if needed and I am happy to support the project if the author has any needs for test hardware.

6by9, 2019-03-25:
https://picamera.readthedocs.io/en/latest/faq.html#preview-flickers-at-full-resolution-on-a-v2-module

```
from picamera import PiCamera

camera = PiCamera()
camera.resolution = camera.MAX_RESOLUTION
camera.start_preview(resolution=(1280, 720)```

mkoppanen, 2019-03-25:
@6by9

I am not using start_preview, as I am rendering the UI in Qt. I found that setting resolution to high values and using capture_continuous with resize parameter results into low FPS (I’m using video port for capture_continuous).

I will get simple code example when I get home but roughly:

1. Initialise camera to resolution and frame rate 
2. Start a thread that loops on capture_continuous and sends each frame using the Qt signal/slot system to main thread 
2.a in the main thread keep rendering incoming frames to QGraphicsView

User has the option to press capture button, which calls “capture” on camera (synchronous in main thread). I have found that if I try to change resolution here the whole pi locks up and needs a hard reboot. If I initialise the camera to preview resolution I get higher fps in preview but at the same time still capture is low res. If I initialise the camera to high resolution and use resize param I get low preview FPS but full res frames.

In my use case flicker on the preview doesn’t matter as I am showing a flash (screen blinks white to indicate capture).

I tried to stop the capture_continuous before changing resolution for a still but this occasionally locks up as well.

6by9, 2019-03-25:
I suspect this is not possible via PiCamera.
The underlying camera component has 3 outputs, nominally designated as preview, video, and stills. Preview and video are normally the same resolution(*), and can operate simultaneously. Capturing a still pauses those pipelines, does the required steps to capture the still, and then resumes preview/video encode.
PiCamera uses the one ```resolution``` parameter to configure both video and stills outputs, therefore I don't believe you have a way to configure them independently. Use of ```resize``` is a software resize, and therefore will struggle when presented with the large stills resolution.

If you switch to using MMAL directly from C then you can set up the pipeline you require.

(*) I seem to recall there is a path that allows them to be different, but compromises the performance on the video encode path.