waveform80/picamera #553: Setting framerate at full resolution v2.1 (3280, 2464)

imageineer, 2019-03-25:
Porting a system from a Raspbian Stretch Lite (2018) to Raspbian Stretch with desktop (updated 25.03.19) I noted that the software raised an error when setting the framerate after setting full resolution for v2.1. Researching the issue I found that it occurred in Python 3.5.3 where I have installed (pip install https://github.com/rwb27/picamera/releases/download/v1.13.1b0/picamera-1.13.1b0-py3-none-any.whl) as well as in 2.7 (picamera that comes with Raspbian). Taking a resolution slightly smaller than full seems OK.  

Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from picamera import PiCamera, PiResolution, mmal, mmalobj
>>> 
>>> c = PiCamera()
>>> 
>>> c.resolution = PiResolution(3280, 2464)
>>> 
>>> c.resolution = PiResolution(1640, 1232)
>>> c.framerate = 10
>>> 
>>> c.resolution = PiResolution(3200, 2404)
>>> c.framerate = 10
>>> 
>>> c.resolution = PiResolution(3280, 2464)
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/pyEnv/lib/python3.5/site-packages/picamera/camera.py", line 2407, in _set_resolution
    self._enable_camera()
  File "/home/pi/pyEnv/lib/python3.5/site-packages/picamera/camera.py", line 1982, in _enable_camera
    self._camera.enable()
  File "/home/pi/pyEnv/lib/python3.5/site-packages/picamera/mmalobj.py", line 734, in enable
    prefix="Failed to enable component")
  File "/home/pi/pyEnv/lib/python3.5/site-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to enable component: Out of resources
>>> 

Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.8 (stretch)
Release:        9.8
Codename:       stretch

Linux 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

kkotowick, 2019-05-08:
Try increasing the memory allocation for the GPU using `sudo raspi-config` -> Advanced -> Memory Split (or similar, can't remember the exact menu names). By default it's set at 128MB, but you might need it at 256MB for full resolution capture.

6by9, 2019-05-08:
Exactly as in the docs https://picamera.readthedocs.io/en/latest/fov.html#hardware-limits
256MB is likely to be excessive, so try 160 or 192MB first and see.