waveform80/picamera #554: Preview in stereoscopic mode

realizator, 2019-03-28:
I use CM3 based solution with 2 CSI Pi cameras, and work in stereoscopic mode.
With any settings of 'stereo_mode' (side-by-side or top-bottom) I succeed in image capture according to all settings. But preview window always shows me only one image from a single camera. I tried to change preview window size, turn on and off 'stereo_decimate', but in any case I can see picture from 1 camera only.
I also noticed, that, for example, for 1280x720 at side-by-side mode I see one image with 640 px width and 720 px height, placed in the middle of the preview window. If I turn off stereoscopic mode, image use all 1280x720 pixels.
If this feature is just not implemented, or am I do something wrong?

6by9, 2019-03-29:
video_render will only render stereoscopic images to a stereoscopic display (eg 3D TV). There is currently no option to render it on a standard screen as SBS or TB.

You (or someone with the same handle) had already asked that question at https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=85012&start=125#p1438275. If the firmware doesn't support it, then PiCamera can't implement it.

realizator, 2019-04-03:
@6by9 Yes, it was me on Pi forum.
It's a pitty that there are no simple way for it. Just a curious, how many users use Pi with 3D TV :-)
And thank you for your answer!