waveform80/picamera #561: capture_sequence with generator causes duplicate images

kkotowick, 2019-05-07:
I'm trying to capture several images with `capture_sequence` while doing some operations in between them. My code looks like so (minimal example):

```
import picamera
import time

def outputs_generator():
    time.sleep(5)
    yield "0.jpg"
    time.sleep(5)
    yield "1.jpg"

if __name__ == "__main__":
    with picamera.PiCamera(resolution=(1640, 1232), framerate=40, sensor_mode=4) as camera:
        time.sleep(3)
        for i in range(1):
            camera.capture_sequence(outputs_generator(), 'jpeg', use_video_port=True)
```

Note that it waits 5 seconds, captures an image ("0.jpg"), waits 5 seconds, and captures a second image ("1.jpg"). The problem is that even though it should wait 5 seconds before taking the second picture, it doesn't seem to be doing so. The function takes 10 seconds to run, as expected, but "1.jpg" is almost identical to "0.jpg" even if I move the camera during those 5 seconds in between `yield`s. It's almost like the camera is capturing two images back-to-back, storing the second one in an internal buffer of some sort, and returning that second one in the second `yield`, but really it should be waiting for that second `yield` before capturing the second image. This seems semi-related to https://github.com/waveform80/picamera/issues/433.

6by9, 2019-05-08:
As expected.

The video port is designed for capturing video. As long as buffers can be found to write into then it will keep taking images. There are typically at least 3 images in the pool to allow for at least double buffering on the display. If you want to slow the image capture rate then you need to amend the frame rate.

The stills port is designed for capturing stills and waits until a trigger parameter is set on the port. I suspect that PiCamera will do that after the output handler completes, therefore it is delayed by your sleeps.

kkotowick, 2019-05-08:
Got it. Is there a way to reduce the pool size, so it only buffers 1 image?

kkotowick, 2019-05-30:
@6by9 Any thoughts on how to reduce the pool size so it only buffers 1 image?

6by9, 2019-05-30:
You can't sensibly as the pool is used for preview and many other things.
If you really want one frame every N, then encode to MJPEG and split the stream. Each frame is a self-contained JPEG.

kkotowick, 2019-05-31:
@6by9 It's not that I need one frame every N milliseconds or such. It's that I need it to not capture an image until the generator function yields. That is, if I do something like change the lighting, then `yield` the buffer, the image that is returned in the buffer _must_ show that lighting change (not be an image that was buffered from before the change).