waveform80/picamera #565: Bayer de-noise algorithm cannot be disabled

karpawich, 2019-05-16:
The "Hardware" section of the PiCamera documentation describes "several post-processing steps on the frame data" by the GPU's image signal processor:

> Bayer de-noise: This is a noise reduction algorithm run on the frame data while it is still in Bayer format.
> 
> ...
> 
> YUV de-noise: Another noise reduction algorithm, this time with the frame in YUV420 format. See **image_denoise** and **video_denoise**.

As explained, the PiCamera Python interface allows the developer to disable the YUV de-noise operation via **image_denoise** and **video_denoise**. However, the Bayer de-noise operation appears to be a separate operation that *cannot* be controlled via those two settings.

Since I am testing the PiCamera v2.1 module as an [entropy source](https://nvlpubs.nist.gov/nistpubs/SpecialPublications/NIST.SP.800-90B.pdf), I need to maximize image static to maximize image entropy. **Given that the YUV de-noise and Bayer de-noise operations are both performed by the ISP, can the Bayer operation also be disabled via the PiCamera Python interface?**

6by9, 2019-05-16:
You need to add 
```
'bayer-denoise': 1<<10,
```
to https://github.com/waveform80/picamera/blob/master/picamera/camera.py#L334, and then set isp_blocks with 'bayer-denoise'.

Notes as per https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=175711 - don't complain if these settings mess up the capture pipeline.

karpawich, 2019-05-27:
@6by9 I've submitted a Pull Request with the simple change you suggested above. Should this Issue be closed, or left open until the change can be done without modifying the source directly?