waveform80/picamera #570: video record 1920x1080

tianjicaoyuan, 2019-05-24:
I use ffmpeg to push hls stream. this is the code :ffmpeg -f v4l2 -r 25 -s 1920x1080 -i /dev/video0 -c:v h264_omx -b:v 7000k....。by the test ,1920x1080 will have problem: ioctl(VIDIOC_STREAMON): Operation not permitted /dev/video0: Operation not permitted。but 1280x720 can be finished。

tianjicaoyuan, 2019-05-24:
I use B3+, V2  IMX219 camera。

6by9, 2019-05-24:
How is that using the PiCamera Python library? /dev/video0 would be using v4l2.

I suggest you post on the forums - https://www.raspberrypi.org/forums/viewforum.php?f=43

tianjicaoyuan, 2019-05-25:
> How is that using the PiCamera Python library? /dev/video0 would be using v4l2.
> 
> I suggest you post on the forums - https://www.raspberrypi.org/forums/viewforum.php?f=43

ok ,Thank you !

closed 2019-05-25 by tianjicaoyuan