waveform80/picamera #572: sensor_mode argument functionality

catrueeb, 2019-06-10:
When initializing a PiCamera intstance using _sensor_mode_, the resolution is not set appropriately.
Example:

```
from picamera import PiCamera
cam = PiCamera(sensor_mode = 4)
print(cam.resolution)
```

Output:
`PiResolution(width = 1920,height = 1080)`

But on sensormode 4 the V2 should set to 1640x1232 resolution. Am I missing something?

6by9, 2019-06-11:
https://picamera.readthedocs.io/en/latest/fov.html

```sensor_mode``` sets the mode on the sensor.

You have the ISP (Image Sensor Pipeline) between the sensor and your application, and that can up or down scale and crop as requested. ```resolution``` sets the resolution at the output of the ISP.

It is perfectly valid to request ```sensor_mode``` 2 for 5 or 8MPix off the sensor, and then resolution of 640x480, it'll just consume a lot more memory in receiving the images from the sensor, and clip your maximum framerate to 15fps.

Your code has chosen to override the sensor mode, therefore the ISP will be taking 1640x1232 in from the sensor, cropping to 1640x922 to match the 16:9 aspect ratio of the output, and then upscaling that to 1920x1080. Again a perfectly valid pipeline.

catrueeb, 2019-06-11:
Thank you @6by9 . I felt that the default output resolution should change to 1640x922 when overwriting the sensor mode with` sensor_mode=4`, but that is therefore a misunderstanding.

closed 2019-06-11 by catrueeb