waveform80/picamera #575: Camera selection

karlahaiat, 2019-07-09:
Hello, 

I was wondering if a camera selection option exists for picamera. In raspivid the camselect option exists so if there are two camera inputs on the board (in my case I am using a stereopi) you can select which camera to use.

6by9, 2019-07-09:
Please read the docs - https://picamera.readthedocs.io/en/latest/api_camera.html#picamera
```
 class picamera.PiCamera(*, camera_num=0, stereo_mode='none', stereo_decimate=False, resolution=None, framerate=None, sensor_mode=0, led_pin=None, clock_mode='reset', framerate_range=None, isp_blocks=None)
```
> Upon construction, this class initializes the camera. The camera_num parameter (which defaults to 0) selects the camera module that the instance will represent. Only the Raspberry Pi compute module currently supports more than one camera.

closed 2019-07-09 by karlahaiat

karlahaiat, 2019-07-09:
For anyone that may be looking for this answer, when I tried replacing camera_num with 1, I got the error: 

picamera.exc.PiCameraError: Unable to determine default GPIO LED pin for RPi revision 3 and camera num 1

This is because in the docs it says: 
> The led_pin parameter can be used to specify the GPIO pin which should be used to control the camera’s LED via the led attribute. If this is not specified, it should default to the correct value for your Pi platform. You should only need to specify this parameter if you are using a custom DeviceTree blob (this is only typical on the Compute Module platform).

RPI.GPIO seems to be returning 3, so just set led_pin parameter to 3 and the camera should work.