waveform80/picamera #577: Purple image captured. Hardware problem?

FutureSharks, 2019-07-27:
I'm running motion detection using picamera and OpenCV. Recently sometimes when the room is dark I get motion detected and the image looks like this:

[image]

I can show the code if need be but is this indicative of a hardware problem?

mfraser, 2019-07-27:
Had this the other day after an update without motion detection running. Went back to an old 1.3 version camera module and it went back to normal so I assumed it was a hardware issue.

Maybe I was wrong?

[image]

barrypj, 2019-07-30:
I did an sudo apt update and sudo apt upgrade and I'm seeing the same thing.  Once it gets dark, my camera starts to show this purple (but also sometimes red) image. As I'm performing a calculation to determine if two adjacent images have changed, every one of them indicates the change as the current purple image is always randomly different from the previous one.  What did you mean by "went back to an old 1.3 version camera"... did you adjust your software or swap out hardware?  Thanks.

barrypj, 2019-07-30:
BTW: Here are some example (problem) images.
[image]
[image]

FutureSharks, 2019-07-30:
There hasn't been a release of picamera in over 2 years so I don't think it's related to picamera unless via later firmware changes.

barrypj, 2019-07-30:
It may be the case that this is an issue for the maintainers of the underlying camera system, as opposed to Python's picamera interface.  I'll see what I can dig up.

mfraser, 2019-07-30:
See this https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=245994

FutureSharks, 2019-07-30:
> To compensate for this purple tinge, there is a new AWB mode

Hmmm, how do we set this in picamera?

barrypj, 2019-07-30:
Thanks for that.  

The 'greyworld' awb mode doesn't appear to be supported in Python's picamera, but there's a workaround here which I'm going to try later: https://github.com/raspberrypi/firmware/issues/1167 - it looks like a bunch of people are working on solutions (with varying degrees of success).

barrypj, 2019-07-30:
I've added this line of code to my Python script:

    cam.awb_mode = 'off'

right after I create a PiCamera() object called cam.  It's not dark here yet (Ireland), so I'm simulating darkness and it appears to have made a difference.  I'm not getting unwanted purple/red images, but won't know if this has fixed things until this evening when it actually gets dark.  I'm also not sure what other problems switching this mode to 'off' will result in, as it is set to 'auto' by default.  We shall see...  ;-)

FutureSharks, 2019-10-29:
I still get problems with this. Seems to be related to running in the dark.

[image]

[image]

Any ideas?

barrypj, 2019-11-19:
I just did a "sudo apt update" and "sudo apt upgrade" on all my PIs and the camera started recording totally black images.  Removing the cam.awb_mode = 'off' line from my scripts has fixed this, and the images are being created fine.  Won't know until tomorrow if anything weird happens at night (and I'll report back then).

FutureSharks, 2019-12-30:
I will close this then. I think it is or was a firmware issue.

https://github.com/raspberrypi/firmware/issues/1229
https://github.com/raspberrypi/firmware/issues/1167
https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=245994
https://raspberrypi.stackexchange.com/questions/104580/picamera-intermitent-purple-snowstorm-emi-or-software-issue

closed 2019-12-30 by FutureSharks

JessicaWoods03, 2020-01-07:
Guys I have a pi 3B+ was using a pi4 but found very little software support when they released it- my pi noir camera turned purple so I did the sudo thingy to fix it a while ago- which worked- temporarily- ummm it turned purple again LOL-