waveform80/picamera #578: Unable to determine if this system is a Raspberry Pi

tmdag, 2019-07-27:
I have installed official supported Ubuntu server from official raspberry Pi website and their guidance (https://ubuntu.com/download/iot/raspberry-pi-2-3)

Now, I'm trying to install picamera from pip but I'm getting an error:
Unable to determine if this system is a Raspberry Pi

 uname -r
4.15.0-1031-raspi2

>>> os.uname()
('Linux', 'raspberrya.hostname.com', '4.15.0-1031-raspi2', '#33-Ubuntu SMP PREEMPT Wed Jan 16 09:52:45 UTC 2019', 'aarch64')

```
Collecting picamera
  Downloading https://files.pythonhosted.org/packages/79/c4/80afe871d82ab1d5c9d8f0c0258228a8a0ed96db07a78ef17e7fba12fda8/picamera-1.13.tar.gz (143kB)
    100% |████████████████████████████████| 153kB 1.6MB/s 
Installing collected packages: picamera
  Running setup.py install for picamera ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-MabEAM/picamera/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-gIAUi4-record/install-record.txt --single-version-externally-managed --compile:
    running install
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-MabEAM/picamera/setup.py", line 145, in <module>
        main()
      File "/tmp/pip-build-MabEAM/picamera/setup.py", line 140, in main
        cmdclass             = {'install': CustomInstallCommand},
      File "/usr/lib/python2.7/dist-packages/setuptools/__init__.py", line 129, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python2.7/distutils/core.py", line 151, in setup
        dist.run_commands()
      File "/usr/lib/python2.7/distutils/dist.py", line 953, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
        cmd_obj.run()
      File "/tmp/pip-build-MabEAM/picamera/setup.py", line 111, in run
        raise ValueError('Unable to determine if this system is a Raspberry Pi')
    ValueError: Unable to determine if this system is a Raspberry Pi

```

clorteau, 2019-08-22:
Same exact issue here. Fixed by doing the following:

`sudo -s
export READTHEDOCS=True
pip install picamera`

6by9, 2019-08-27:
> I have installed official supported Ubuntu server from an official raspberry Pi website and their official guidance (https://www.raspberrypi.org/downloads/).

Minor point, Ubuntu Server is listed under "Third Party Operating System Images", with the comment
"Third-party operating system images for Raspberry Pi are also available:".
3rd party means all support is by third parties (in this case Canonical). It is not downloaded from the official Raspberry Pi website, nor is it supported by Raspberry Pi (Trading) Ltd.

HughWarrington, 2020-06-15:
This seems to have been fixed in master in c145015. But not yet in a release -- the most recent of which was more than three years ago. I can't work out if this project is still maintained, or maybe there is just no desire to do another release?

raulgool, 2020-07-31:
Thank you @clorteau 
Your reply did exactly what I wanted to achieve.
After doing this, I had to 2 more things to get `import picamera` working without any errors.

**Background:**
Hardware: Raspberry Pi 4 model B. 4 GB version
OS: ubuntu-20.04-preinstalled-server-arm64+raspi.img.xz

**Extra Steps:**
  1.  Install colorzero library. Use pip or pip3 based on python2 or python3
 `sudo pip install colorzero`
  2. Copy firmware libraries from https://github.com/raspberrypi/firmware/tree/master/opt/vc/lib here to \usr\lib\
 I maybe wrong regarding copying all the files but I was too impatient to go through each of them one by one.

**Testing**
 I tried basic code from https://picamera.readthedocs.io/en/release-1.13/recipes1.html#capturing-to-a-file. It worked only when I ran it as `sudo python3 test.py`.
```python3
#test.py
from time import sleep
from picamera import PiCamera

camera = PiCamera()
camera.resolution = (1024, 768)
camera.start_preview()
# Camera warm-up time
sleep(2)
camera.capture('foo.jpg')
```

heike380, 2020-08-20:
@raulgool :
How did you manage to run the test.py exactly?

I always get the error:
`OSError: libbcm_host.so: wrong ELF class: ELFCLASS32`

Probably meaning, that the libbcm_host.so library used from picamera is not built for 64 bit.

I have the same setup like you:
Hardware: Raspberry Pi 4 model B. 4 GB version
OS: ubuntu-20.04-preinstalled-server-arm64+raspi.img

I installed picamera (worked without issues):
`pip3 install picamera`
and colorezo:
'pip3 install colorezo'

I copied all the lib* files from https://github.com/raspberrypi/firmware/tree/master/opt/vc/lib to /usr/lib on the Pi and added the path to LD_LIBRARY_PATH.

raulgool, 2020-10-09:
@heike380 I am not sure if you figured out the issue or not. But it seems like the repository is updated since I last downloaded the files. I don't remember when I downloaded the files
Try with files from two following commits:

https://github.com/raspberrypi/firmware/tree/2b76cfc6f57d4943144b9ceb5b57d3d455d6a8fd/opt/vc/lib
https://github.com/raspberrypi/firmware/tree/542aceb30364db3ce87d532ea90451f46bbb84e9/opt/vc/lib