waveform80/picamera #586: Preview doesn't display with HDMI monitor

Devilmachie, 2019-08-27:
Hello,

I have tried to begin with the picamera module and i started with simple scripts such as :      

`from time import sleep
import picamera

camera = picamera.PiCamera()
camera.start_preview()
camera.start_recording('/home/pi/Desktop/video.h264')
sleep(5)
camera.stop_recording()
camera.stop_preview()`

The video is well registered and the script doesn't output anykind of error. The problem is that the method start_preview doesn't display anything on my screen. Is there any common explanation for this?

6by9, 2019-08-28:
Have you enabled the full GL driver? Check for "dtoverlay=vc4-kms-v3d" in /boot/config.txt.

DRM/KMS does not allow multiple rendering clients, and is also totally controller by the Linux kernel, therefore the GPU can not add an extra overlay layer for the preview.

ADFRVH, 2020-12-07:
And so how do we solve the problem?

6by9, 2020-12-08:
> And so how do we solve the problem?

You get the OP to answer the question of whether they enabled the full GL driver (in which case you solve it by disabling it to use FKMS or legacy mode), or otherwise provide some diagnostics.

optimuspaul, 2020-12-14:
I have `dtoverlay=vc4-kms-v3d` what should it be? @6by9 isn't very clear in remediation, please advise.

6by9, 2020-12-15:
Remove ```dtoverlay=vc4-kms-v3d```, or replace it with ```dtoverlay=vc4-fkms-v3d``` (note the 'f').

Picamera relies on the GPU firmware to be able to add overlays. vc4-kms-v3d switches all control of the rendering pipeline (and 3D) to the Linux kernel via the DRM/KMS APIs, so the GPU is no longer able to do that.
vc4-fkms-v3d is a hybrid that emulates the DRM/KMS APIs on top of the firmware controlled rendering pipeline.

optimuspaul, 2020-12-15:
thanks, that worked for me

closed 2023-02-05 by Devilmachie