waveform80/picamera #592: Flexible Picamera.rotation value

MCYBA, 2019-10-06:
Thanks for all this work.
I have trouble with `PiCamera.rotation`. I need to use this variable with values that are not multiples of ninety.

Seems in the PiCamera class the `rotation` enforced to be multiples of ninety. Could you provide a way to use this with more flexible angles?

best regards.

6by9, 2019-10-07:
No.

The processing pipe supports the basic operations of transpose (reflect in x=y), horizontal flip, and vertical flip. These allow you to do all variations of 90 degree rotations, with and without reflection.

There is no hardware within the imaging pipe that allows more complex rotations to be performed.

MCYBA, 2019-10-07:
Got it, thanks for the reply

closed 2019-10-07 by MCYBA