waveform80/picamera #600: Possible issue on RPi CM3+

simonallen5, 2019-11-27:
I've been this excellent lens shading correction procedure successfully on a RPi CM3 compute module, with two cameras attached. However, I've now switched to the new CM3+ - same board and wiring, just swapping the CM3 module out - and now seem to have an issue.

Forked picamera is installed and lens_shading is supported. However, when I try to write to the lens_shading table (as in your snippet below) the camera preview will not show. No error shows though. Run exactly the same code on the CM3 in the same socket and it works.

i.e. with PiCamera as cam:
  cam.start_preview()              works

with PiCamera(lens_shading_table=lst) as cam
  cam;start_preview()            doesn't

6by9, 2019-12-02:
Sorry, I haven't got a CM set up at present.
Can you test using raspistill please so we can identify whether it is firmware or Picamera?

CM3 and CM3+ run identical firmware. They change the revision of the main SoC, and have slightly thicker PCBs to dissipate heat, but are otherwise are identical.
It's not as if lens shading is disabled in your first case, it's just got different numbers in it.

Is this a v1 (OV5647) or v2 (IMX219) camera module that you're using? And what firmware versions?

simonallen5, 2019-12-03:
Thanks. Both cameras on the CM3+ are v2 (IMX219). Not sure what firmware versions - how do I check that (sorry - afraid I'm not a Pi expert!)
The cameras work fine via the raspistill -cs 1 -v -o test.jpg command. How do I check the lens shading via raspistill?

It is definitely something to do with setting the shading table: lens_shading_table=None works OK

6by9, 2019-12-03:
```vcgencmd version``` tells you the firmware version.
To confirm, are these CM3L's (ie external SD card) or the CM3 (internal EMMC)? I'm assuming CM3L as otherwise it is hard to claim you are running exactly the same code on the same system.

The lens_shading branch on https://github.com/6by9/userland/tree/lens_shading was my test code for lens shading in raspistill.
```
git clone https://github.com/6by9/userland
cd userland
./buildme
raspistill -t 0
```
NB This does replace the installed version of raspistill and other apps. You'll want to reinstall the relevant Debian package afterwards to put them back, but I can't remember the package name at present.

simonallen5, 2019-12-03:
Thanks for the quick reply.

Version is 6c3fe3f096a93de3b34252ad98cdccadeb534be2

No, I am using the CM3+ (and previously CM3) EMMC version. I’m sure the program code is the same as I’ve just copied it across, but do recognise that there is a chance (good chance?) that I have done something different in the set up of the system. I certainly can’t be sure that everything was installed in the same order.

I’ll try the your raspistill version now and let you know how I get on

6by9, 2019-12-03:
Note I've updated the test instructions - I'd forgotten to check out the right branch.

simonallen5, 2019-12-03:
The git clone command worked OK but git checkout generates an error: fatal: not a git repository (or any of the parent directories): .git

6by9, 2019-12-03:
Doh, ```cd userland``` first, then ```git checkout -t -b lens_shading origin/lens_shading```

simonallen5, 2019-12-03:
Thanks – I should have worked that out!

Just to check – I need to install cmake and an arm cross compiler before I can build?

One other thought – I set up the CM3+ with Debian Buster whilst the previous implementation on the CM3 was under Stretch – is that likely to make any difference?

Thanks again for your help

6by9, 2019-12-03:
You need cmake, nothing else.

Buster vs Stretch will have changed a whole bundle of things! Please test with the same environment.

6c3fe3f096a93de3b34252ad98cdccadeb534be2 is from July 2019, so that sounds like the first Buster release. It's always worth updating to ensure it's not something that has already been fixed.

simonallen5, 2019-12-03:
OK. Don’t know why I didn’t think of that in the first place. I’ve cloned the Stretch image from the CM3 onto the CM3+ and everything works fine, so either I screwed up my Buster setup on the CM3+ or there is something in Buster it doesn’t like. I will try to upgrade the working version from Stretch to Buster and see what happens.

Many thanks for your help

6by9, 2019-12-03:
Personally I would have tried Buster on the CM3 as Stretch is no longer supported, but knowing that the Stretch image works on both is still a useful data point.

simonallen5, 2019-12-09:
Just to update on this – I have now run a buster update on the CM3+ stretch installation, and reinstalled the forked picamera module. All works fine this time round – I can load custom lens-shading tables on both cameras in the system, and have calibrated them both.

So it must have been something in my setup of buster from scratch on the original CM3+ installation that was causing the problem.

6by9, 2019-12-09:
Please close the issue if the problem is resolved.

closed 2019-12-09 by simonallen5