waveform80/picamera #603: Issue video streaming with 4-camera adapter board v2

mkeyno, 2019-12-24:
Hi
I bought a [Multi-Camera Adapter Module for Raspberry Pi V.2](https://www.arducam.com/multi-camera-adapter-module-raspberry-pi/) for my stereo project. I use gpio multiplexer code along with picamera module to read stream video, but the camera stop working after a couple of second
following is my code and error
any help really appreciated
#code
```python
 import RPi.GPIO as gp
# initialize the camera and grab a reference to the raw camera capture
gp.setwarnings(False)
gp.setmode(gp.BOARD)
gp.setup(7, gp.OUT)
gp.setup(11, gp.OUT)
gp.setup(12, gp.OUT)
gp.output(7, False)
gp.output(11, False);gp.output(12, True)
#########################################
#########################################
from picamera.array import PiRGBArray
from picamera import PiCamera
import cv2
import time

camera = PiCamera()
camera.resolution = (320, 240)
camera.framerate = 30
rawCapture = PiRGBArray(camera, size=(320, 240))
display_window = cv2.namedWindow("Faces")
time.sleep(1)
i=0
for frame in camera.capture_continuous(rawCapture, format="bgr", use_video_port=True):

    image = frame.array
    #DISPLAY TO WINDOW
    cv2.imshow("Faces", image)
    print("image num=",i);i+=1
    key = cv2.waitKey(1)
    rawCapture.truncate(0)
    time.sleep(1)
    if key == 27:
        camera.close()
        cv2.destroyAllWindows()
        break
```
#error
```python
image num= 89
image num= 90
image num= 91
image num= 92
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1704, in capture_continuous
    'Timed out waiting for capture to end')
picamera.exc.PiCameraRuntimeError: Timed out waiting for capture to end

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/Public/camera test.py", line 29, in <module>
    for frame in camera.capture_continuous(rawCapture, format="bgr", use_video_port=True):
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1710, in capture_continuous
    encoder.close()
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 431, in close
    self.stop()
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 419, in stop
    self._close_output()
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 349, in _close_output
    mo.close_stream(output, opened)
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 371, in close_stream
    stream.flush()
  File "/usr/lib/python3/dist-packages/picamera/array.py", line 238, in flush
    self.array = bytes_to_rgb(self.getvalue(), self.size or self.camera.resolution)
  File "/usr/lib/python3/dist-packages/picamera/array.py", line 127, in bytes_to_rgb
    'Incorrect buffer length for resolution %dx%d' % (width, height))
picamera.exc.PiCameraValueError: Incorrect buffer length for resolution 320x240 
```

6by9, 2019-12-25:
If it works with a single camera connected directly to the Pi, then please raise this with the manufacturer of your multiplexer board (Arducam).

mkeyno, 2019-12-27:
@6by9 the board perfectly work if not continuously take an image, it seems to callback handler not return in timeout , and thats make to chock the thread

6by9, 2019-12-27:
If continuously taking a capture without the board works, then it's still an issue with the board.

CSI2 has per line checksums and error correction. IIRC it can cope with a 1 bit error per line, and 2 or more marks the frame as invalid and will result in them being discarded under some conditions. Multi-bit errors are not expected with standard connections, therefore it sounds like the mux board is introducing signal integrity issues. Again, not a Pi or a PiCamera issue.

minjianmj123, 2020-04-13:
Hi, im using arducam multi camera adapter for my project as well. Im only able to on camera A. When camera A switch to B and C error occurred.
the error is: [ WARN:0] global ../opencv/modules/videoio/src/cap_v4l.cpp (998) tryIoctl VIDEOIO(V4L2:/dev/video0): select() timeout.

Have you guys faced this issue?

Thank you.

6by9, 2020-04-13:
@minjianmj123 If you're getting an error relating to V4L2 and /dev/videoX, then you're not using Picamera (this library). Please post your question in a sensible place.

minjianmj123, 2020-04-13:
Hi, I would like to ask you guys about problems faced when using 4-camera adapter board v2 since you guys have been using fot past projects. That's why I posted my question here.

Thank you.

mkeyno, 2020-04-25:
@minjianmj123  the hardware design has a flaw, no flow control considered for a stream, and that's make choking the bus line