waveform80/picamera #604: Python 3.8.1 TypeError : vc_dispmanx_element_add.argtypes = [DISPMANX_UPDATE_HANDLE_T, DISPMANX_DISPLAY_HANDLE_T, ct.c_int32, ct.POINTER(VC_RECT_T), DISPMANX_RESOURCE_HANDLE_T, ct.POINTER(VC_RECT_T), DISPMANX_PROTECTION_T, VC_DISPMANX_ALPHA_T, DISPMANX_CLAMP_T, DISPMANX_TRANSFORM_T] TypeError: item 9 in _argtypes_ passes a union by value, which is unsupported.

jmmec, 2019-12-27:
Hello,

picamera works great with Python 3.8.0, however a TypeError is immediately raised upon "import picamera" in Python 3.8.1 as shown below:

```
pi@pi1:~/DEV/CVDEV $ ./cvStart
Traceback (most recent call last):
  File "./jmTest.py", line 16, in <module>
    import picamera
  File "/home/pi/DEV/VENV_OPENCV_PYTHON3_8/lib/python3.8/site-packages/picamera/__init__.py", line 72, in <module>
    from picamera.exc import (
  File "/home/pi/DEV/VENV_OPENCV_PYTHON3_8/lib/python3.8/site-packages/picamera/exc.py", line 41, in <module>
    import picamera.mmal as mmal
  File "/home/pi/DEV/VENV_OPENCV_PYTHON3_8/lib/python3.8/site-packages/picamera/mmal.py", line 47, in <module>
    from .bcm_host import VCOS_UNSIGNED
  File "/home/pi/DEV/VENV_OPENCV_PYTHON3_8/lib/python3.8/site-packages/picamera/bcm_host.py", line 451, in <module>
    vc_dispmanx_element_add.argtypes = [DISPMANX_UPDATE_HANDLE_T, DISPMANX_DISPLAY_HANDLE_T, ct.c_int32, ct.POINTER(VC_RECT_T), DISPMANX_RESOURCE_HANDLE_T, ct.POINTER(VC_RECT_T), DISPMANX_PROTECTION_T, VC_DISPMANX_ALPHA_T, DISPMANX_CLAMP_T, DISPMANX_TRANSFORM_T]
TypeError: item 9 in _argtypes_ passes a union by value, which is unsupported.
```

DX37, 2019-12-30:
I have this issue too. Just downgraded to 3.8.0 and it's fine.

wingkitlee0, 2019-12-30:
I also had this issue (self-compiled python 3.7.6). I ended up reflashing the OS raspbian to the newest version. It works fine now with python 3.7.4 . It seems to be a library issue than a python one..

hsiejkowski, 2019-12-30:
@wingkitlee0 I confirm your suspicion. I tested it with Python 3.8.1 and 3.7.6 with the same error. In my case, the error started to happen after a system upgrade.

laudge, 2019-12-30:
same problem for me with python 3.8.1 .  bcm_host.py seams to be the cause. But I do not know how to solve it. I will revert to previous python version.

riedwaanb, 2019-12-31:
Also confirming this issue with self compiled 3.8.1.

moaxey, 2020-01-11:
I experienced this error in python 3.8.0 and 3.8.1 installed with pyenv and picamera installed with pip or pipenv..
After installing [python 3.8.0 from source following this method](https://installvirtual.com/how-to-install-python-3-8-on-raspberry-pi-raspbian/) (but install not altinstall), and no pyenv lines in my .bashrc. 
Then I could `pip3 install picamera` and import it without the error mentioned above.

BlackLight, 2020-01-21:
This has been a [long-standing issue](https://bugs.python.org/issue16575) in the underlying interpreter. Basically, the interpreter's support for `ctype.Union` passed by value has always been quite limited, as it's hard for the interpreter to get the right memory size, and it's silently been dropped in Python 3.8.1.

I was investigating how to replace `DISPMANX_CLAMP_KEYS_T` with a `Structure` or an array, but I've eventually found out that the incriminated commit has [luckily been reverted](https://github.com/python/cpython/pull/17960) after it broke quite a lot of things. At this point the best solution is to either downgrade to 3.8.0 or wait for 3.8.2.

sgnyjohn, 2020-03-02:
nunca trabalhei com linguagem tão instável, não é confiável.

nickmmain, 2020-03-24:
Can confirm that going from 3.6.7 to 3.7.7 solved this issue for me.

hsiejkowski, 2020-03-26:
With python 3.8.2 it works too.