waveform80/picamera #606: How to launch 2 side-by-side preview windows?

PaladinEng, 2020-01-04:
I'm trying to create two side-by-side preview panes, split from the same camera input, and use ROI changes to create a zoom effect., as described [here:](https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=259241&p=1589079#p1589079)

I've approached this from two angles. One, through the basic picamera recipe, allows me to get a zoom by using camera.zoom(), and I can go from the full camera output to a reasonable zoom,  but I have not figured out how to create a second preview window.

In the second approach, I'm using MMAL and the MMALsplitter to get the video presentation that I want, but I have not been able to figure out a dynamic zoom. 

So, in the first case, how can I create a second preview window?
In the second case, how can I vary the zoom by cropping the camera output?

6by9, 2020-01-09:
The first case is going to be a dead end - the camera will only create a single crop output.

If you take the camera producing full FOV and feed it into a splitter. Take two splitter outputs and feed them into two render components. Set the source rectangle (it looks like PiCamera is exposing this as [crop](https://picamera.readthedocs.io/en/latest/api_renderers.html#picamera.PiRenderer.crop). Then you should end up with the result you want. Sorry, I don't do Python, therefore I can't help you on the specifics.

mikevanis, 2020-01-29:
A third approach would be to grab the image array with picamera, and then manipulate it with OpenCV. You could start off with a big image and extract the ROI, and use imshow to write it into one window, then take the same image, resize it and use another imshow to write it into a second window.