waveform80/picamera #607: picamera.exc.PiCameraMMALError: no buffers available: Resource temporarily unavailable; try again later

smarti57, 2020-01-06:
I'm unable to run Crosshair example from Chap 16 without buffer issues on an RPI4 with the latest raspbian buster and both the binary version of picamera and a compiled version from the latest master.  I do get frames that update in a non-periodic manner (maybe 0.2fps on average) 

When running the example in both python 2.7.16 and python 3.7.3 I get the following against the latest version of picamera master,
```
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 232, in 'calling callback function'
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 1227, in wrapper
    self._pool.send_buffer(block=False)
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 1931, in send_buffer
    super(MMALPortPool, self).send_buffer(port, block, timeout)
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 1881, in send_buffer
    raise PiCameraMMALError(mmal.MMAL_EAGAIN, 'no buffers available')
picamera.exc.PiCameraMMALError: no buffers available: Resource temporarily unavailable; try again later
```

smarti57, 2020-01-06:
I tried the monkey patch from here,

[(https://github.com/dtreskunov/rpi-sensorium/commit/40c6f3646931bf0735c5fe4579fa89947e96aed7#diff-0d6ec65fb5e525684e82dca730c48341)]

but have issues as I'm not using a picamera object, but rather MMALobjs.

DdraigX, 2020-04-14:
@smarti57 what do you have to do with the patch to enable it? I am very green to python etc so any help would be appreciated. Currently my little raspberry pi project is working but getting this error.