waveform80/picamera #608: Stereoscopic image capture at max_resolution fails

realizator, 2020-02-06:
I have a Compute Module-based stereoscopic setup with the two V1 cameras.
I'm trying to capture stereoscopic image (sbs) with the maximum sensor resolution and without decimate (i.e. 5184x1944) and got an error:
"Invalid resolution requested: %r" % (value,)"

The same with the 5120x1936 (height is multiple of 16, width is multiple of 128 for each camera).

But with the _raspistill_ I can successfully capture stereoscopic image with these options. If it is possible to capture such an image with the PiCamera from Python?

6by9, 2020-02-06:
The test at https://github.com/waveform80/picamera/blob/master/picamera/camera.py#L2609 wants to account for stereoscopic.

realizator, 2020-02-06:
Quick check with disabling this test leads to other errors (in self._enable_camera(), self._camera.enable(), raise PiCameraMMALError and finally "Failed to enable component: Out of resources"). Looks like this _max_resolution_ size test is used in other places, and simple trick I've try is not the way here.

6by9, 2020-02-06:
Most likely you'll need to increase max_still_width and max_preview_width (assuming SBS rather than TB stereo mode) at https://github.com/waveform80/picamera/blob/master/picamera/camera.py#L2303-L2307.
Increase the height if using top/bottom, although it probably doesn't make much difference.

realizator, 2020-02-06:
Ok, thank you, I will check this!

realizator, 2020-02-07:
Well, your advice from https://github.com/waveform80/picamera/issues/608#issuecomment-582952672 works great, if I increase GPU_MEM in config.txt to 256 Mb. No any other code changes needed.
With GPU_MEM 192Mb I get "Out of resources".
Usually I use 128 Mb, and raspistill works fine. But looks like PiCamera needs more.

realizator, 2020-02-18:
@6by9 I also have a bunch of other stereoscopic support questions (for example top-bottom is broken, rotation is broken etc). I find a lot of comments in PiCamera documents, that the stereoscopic support is not fully tested, and also that latest 1.13 release was on Feb, 2017. 
I can help with the stereoscopic hardware, and also give a description of a found bugs. Is @waveform80 has a time and interest to update stereoscopic support in PiCamera?