waveform80/picamera #612: Picamera.camera.capture documentation: Broken link for "definitions of quality"

luxedo, 2020-02-21:
In the [picamera documentation for the camera object](https://picamera.readthedocs.io/en/release-1.10/api_camera.html#module-picamera.camera) there is a link in the `quality` section and it's broken:

> quality - Defines the quality of the JPEG encoder as an integer ranging from 1 to 100. Defaults to 85. Please note that JPEG quality is not a percentage and [definitions of quality](https://www.photo.net/learn/jpeg/#qual) vary widely.

I searched for the article and couldn't find it. I have no idea how was the original article, any suggestions?

6by9, 2020-02-21:
The Wayback Machine is your friend
https://web.archive.org/web/20161221193944/http://photo.net/learn/jpeg//#qual

luxedo, 2020-02-21:
Thanks a lot! I submitted a for updating this link

closed 2020-02-21 by luxedo