waveform80/picamera #614: PiCamera.camera reduced quality range

luxedo, 2020-02-21:
I feel that the camera quality is not working as intended, at least in the v2 camera.

I tested several image quality settings:
| file_name      | rpi_quality | ident_quality | size (KB) |
|----------------|-------------|---------------|-----------|
| q_001_0001.jpg | 1           | 1             | 152       |
| q_010_0001.jpg | 10          | 75            | 852       |
| q_020_0001.jpg | 20          | 94            | 2000      |
| q_030_0001.jpg | 30          | 98            | 3144      |
| q_040_0001.jpg | 40          | 99            | 3700      |
| q_050_0001.jpg | 50          | 99            | 4260      |
| q_060_0001.jpg | 60          | 100           | 4940      |
| q_070_0001.jpg | 70          | 100           | 4940      |
| q_080_0001.jpg | 80          | 100           | 4944      |
| q_090_0001.jpg | 90          | 100           | 4944      |
| q_100_0001.jpg | 100         | 100           | 4940      |

The second column is the quality argument for `camera.capture` and the third is the quality from [ImageMakick's](https://imagemagick.org/index.php) `identify` command. 

We can see that the quality and the image size changes up to around quality 50 for the RPi, and then it stays the same. I'm guessing this is not the intended behavior. Any thoughts?

luxedo, 2020-02-21:
The curve that fits these data is something like:

__ident_quality = 100 - 100 / (rpi_quality^2/25)__

This curve don't fit well the data below 10.

6by9, 2020-02-21:
There is no defined mapping of quality for JPEGs in the spec - I queried it with the JPEG expert whilst at Broadcom 6+ years ago.
The fact that the quality factor doesn't match that used by ImageMajick is just one of those things.

luxedo, 2020-02-21:
I think that the `quality` argument is kind of confusing, specially the fact that the range between 60-100 that doesn't even affect the image size. 

Do you think it would be interesting to reduce the range to 1-60? Or to map the rpi quality to imagemagic to make the range 0-100 more intuitive?

6by9, 2020-02-26:
Two major issues:
1) working out exactly how the hardware setups up the encoder quantisation coefficients. Those who designed the hardware aren't around to ask, so it's not a trivial task.
2) backwards compatibility. If someone is currently using eg ```quality=30``` and an update changes the quality of the output on them, we get complaints.

luxedo, 2020-02-26:
I see, indeed it's more complicated than I thought. If you ever think about messing with this in the future I'd be glad to help.

I think I'm going to make a more detailed map of the qualities and post it here as reference.

closed 2020-02-26 by luxedo

luxedo, 2020-04-25:
I did another test sweeping through all values for the quality argument. As we can see the Imagemagick's algorithm for computing quality gets confused in the lower portion of the graph, otherwise the following approximation might work:
<img src="https://render.githubusercontent.com/render/math?math=$ y \approx \dfrac{100 {x}}{(1 - x)}$">
<img src="https://render.githubusercontent.com/render/math?math=$x = \dfrac{pc^2}{25}$">

[image]

https://gist.github.com/luxedo/c954a13618830b7aeef2cae1e040a286