waveform80/picamera #617: PiCamera.colorspace does not exist

ExecDevOps, 2020-04-11:
Hi,

While trying to better the image quality from PiCamera's "start_recording" function I came across the "colorspace" attribute listed in the manual. It states that using 'jfif' table for MJPEG video will increase the luma range, which as I understand can make quite a difference.

I was however not successful in implementing it, every attempt resulted in the error message "AttributeError: 'PiCamera' object has no attribute 'colorspace'". Looking at the "picamera" Python source revealed that it is indeed not implemented, contrary to the manual. 

Searching the 'net lead me to issue #512 (https://github.com/waveform80/picamera/issues/512) where this was mentioned to be resolved for release 1.14.

My question is if I could get a pointer as to where/how hardcode this setting until 1.14 gets released, or any other way of changing the colorspace to 'jfif'.

Many thanks for a great product and thank you in advance for any help with this.

6by9, 2020-04-12:
colourspace only modifies the signalling, not the actual image data (I'd been asking for it to actually do that for a long time, but never mind).
H264 has signalling in the headers to denote BT601 or BT709 primaries, and full or limited range. JFIF is full range with the same primaries as BT601.
MJPEG doesn't have such signalling, so it'll do nothing.

ExecDevOps, 2020-04-12:
Thank you for quick response, the docu is not clear about MJPEG not being affected but now I know better :)

Have a great Easter!

dans98, 2020-04-13:
the latest docs suggest jfif for mjpeg.

https://picamera.readthedocs.io/en/latest/api_camera.html?highlight=jfif

> However, when recording MJPEG video, you may want to use the “jfif” table instead as it produces luma values in the 0-255 range, rather than the 16-235 range produced by the standard tables.