waveform80/picamera #625: Strange frame index behaviour

mikhailkin, 2020-05-17:
I've faced with quite odd frame index numbers coming from picamera module. The indices differs from overlay produced by camera.annotate_frame_num = True

Code to reproduce:
```
import picamera
import picamera.array
import numpy as np

class MyMotionDetector(picamera.array.PiMotionAnalysis):
    def analyse(self, a):
        print(f"{self.camera.frame.index}")

with picamera.PiCamera() as camera:
    camera.resolution = (640, 480)
    camera.framerate = 30
    camera.start_recording(
        '/dev/null', format='h264',
        motion_output=MyMotionDetector(camera)
        )
    camera.wait_recording(30)
    camera.stop_recording()
```

Output:

> 2
> 4
> 6
> 8
> 10
> 12
> 14
> 16
> 18
> 20

During long run indices might diverge quite greatly (not by factor of 2)

RDReavis, 2020-05-24:
The frame index is incremented when:
1. Video frame data is passed to a normal output (/dev/null, in your case)
2. Video motion estimation data is passed to a motion detector

This is likely why you're seeing the frame index incrementing by two.

The divergences you're seeing might be attributed to SPS headers being inserted at regular intervals, but I'm not certain.

mikhailkin, 2020-05-25:
@RDReavis, thank you for the clue.
Still, at the end we have a few different values for frame index at the end points. Any reasons to expect it? Shall we multiply dependent almost equal entities?